wills106/homeassistant-solax-modbus #1506: [Bug]: 'Inverter Brand' - 'Inverter Model' Description of Error

michagro, 2025-07-29:
### Describe the bug

Same Error which in HACS Solax  integration 2025.07.04

sensor.growatt_pv_today_s_solar_energy Shows no kWh 

In Version 2025.07.06 all ok but in current Version no value 

### Integration Version

2025.07.07

### Homeassistant core version

2025.07.04

### Inverter brand

Growatt SPH 6000 BH-UP TL 3 

### Plugin used

plugin_growatt.py

### Serial prefix

YBAA07

### Connection Method

Waveshare RS485 to ETH

infradom, 2025-07-29:
> There is more stuff missing, l

I believe this complaint is different and should be solved with today's 2027.7.7 release. Ary you sure you are running that version? If so, also open the failing entities configureation menu, to see if the key did not receive an unwanted _2 suffix. If so, remove the _2 suffix on the failing control.

spetrie13, 2025-07-29:
Solis hybrid inverter

Latest has broken the battery in out that was fixed in  .6 version.

TheEragon, 2025-07-29:
> > There is more stuff missing, l
> 
> I believe this complaint is different and should be solved with today's 2025.7.7 release. Ary you sure you are running that version? If so, also open the failing entities configureation menu, to see if the key did not receive an unwanted _2 suffix. If so, remove the _2 suffix on the failing control. 
> 
> See also closed issue #1496 

Nope I don't see any _2

infradom, 2025-07-29:
> Nope I don't see any _2

Are you sure you are running verson 2025.7.7 ?

flopp999, 2025-07-30:
> There is more stuff missing, like almost every option under Configuration, all Battery First sliders, etc. Everything broke in 2025.07.04 and afterwards
> 
> <img alt="Image" width="559" height="178" src="https://private-user-images.githubusercontent.com/5801389/472171648-76128fb6-4c77-4115-ac86-690920f1a56b.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NTM4NjU2NDEsIm5iZiI6MTc1Mzg2NTM0MSwicGF0aCI6Ii81ODAxMzg5LzQ3MjE3MTY0OC03NjEyOGZiNi00Yzc3LTQxMTUtYWM4Ni02OTA5MjBmMWE1NmIucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI1MDczMCUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNTA3MzBUMDg0OTAxWiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9YWRlZjIzNWI3ODA3MzAzYzRlYTlkMDA1ODY0YmI4ZjczYjc4ODY0NDk0YTY5ODhjNTFjZjdlNDQzMjYxNzFhNSZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QifQ.DubKhgWQ_q3aeWl-6NCa6CeXijXLHBM1wn5G9PwhSSM"> <img alt="Image" width="590" height="464" src="https://private-user-images.githubusercontent.com/5801389/472171680-5cc8f9ce-5c60-4081-ad84-0a245eb53658.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NTM4NjU2NDEsIm5iZiI6MTc1Mzg2NTM0MSwicGF0aCI6Ii81ODAxMzg5LzQ3MjE3MTY4MC01Y2M4ZjljZS01YzYwLTQwODEtYWQ4NC0wYTI0NWViNTM2NTgucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI1MDczMCUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNTA3MzBUMDg0OTAxWiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9Y2Q0MDYwNWI2N2ZlMjBkZGYxMGJlN2M0MDMwNjk2MzU5MDI2MzA2MzE1MmMzYjNkYWI0MDdiNGMxZjNkNTEzZiZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QifQ.Pgagh06v9NvpKBvCfnjAd8TIyiv7XAo4G8uZqUnFxXc">
> [@flopp999](https://github.com/flopp999) would you mind checking out what's wrong please?

Hi, I have a GEN4 and that entity is for GEN3, so I cannot test it.
It is not changed by me, if something has been changed

flopp999, 2025-07-30:
> Solis hybrid inverter
> 
> Latest has broken the battery in out that was fixed in .6 version.

This issue is about Growatt inverter, please create new issue if you still have problem

flopp999, 2025-07-30:
> I'm sure since I tried to upgrade to 2025.7.7 and had to revert 2025.7.3 because almost everything was broken, including the `sensor.growatt_pv_today_s_solar_energy`
> 

For me it is working fine with `sensor.growatt_pv_today_s_solar_energy`
I have 15KTL3-XH, GEN4

felixhix, 2025-08-06:
Have the same effect since update. Entity "Today's Solar Energy" is always 0 kWh. Have a Growatt SPH 10k TL3-BH UP.

Laksepote, 2025-08-07:
Same problem here with Growatt SPH-3600. No data for todays solar production

[image]

brad5505, 2025-08-09:
Also SPH 5000TL-HUB has the same problem. No sensor.growatt_pv_today_s_solar_energy. Reveted back to version 2025.7..

flopp999, 2025-08-09:
So with 2025.7.7 GEN3 do not get Todays Solar Energy?
It works with 2025.7.6?

Can anyone with 2025.7.7 upload a log? maybe there can be an answer why it is not working

flopp999, 2025-08-10:
I can try to solve this but lets see if anyone with a GEN3 can give me logs and help me.

flopp999, 2025-08-11:
@TheEragon @felixhix @brad5505 @Laksepote @michagro 
Noone that can help to make this solved?

flopp999, 2025-08-11:
@Laksepote OK my mistake.

can you try with these settings in modbus.yaml

maybe you neeed to change port and baudrate depending of your settings in HA and inverter

```
 - name: Growatt
    type: serial
    method: rtu
    port: /dev/ttyUSB0
    baudrate: 9600
    stopbits: 1
    bytesize: 8
    parity: N
    sensors:
      - name: PV1 test
        unique_id: pv1test
        slave: 1
        address: 59
        input_type: input
        data_type: uint32
        scan_interval: 60
```

Laksepote, 2025-08-11:
Error loading /config/configuration.yaml: mapping values are not allowed here
  in "/config/modbus.yaml", line 2, column 9

Laksepote, 2025-08-11:
` - name: Growatt
    type: serial
    method: rtu
    port: /dev/serial/by-id/usb-1a86_USB_Serial-if00-port0
    baudrate: 9600
    stopbits: 1
    bytesize: 8
    parity: N
    sensors:
      - name: PV1 test
        unique_id: pv1test
        slave: 1
        address: 59
        input_type: input
        data_type: uint32
        scan_interval: 60`

Laksepote, 2025-08-11:
That worked ;-)

PV1 test value is "102"

flopp999, 2025-08-11:
Great. This means the problem is not to get the value from the inverter.
Let me check more in the code what can cause this problem for those with GEN3.

flopp999, 2025-08-11:
@Laksepote can you change the register/address to 3083.
Name and everything else can be same as before.

For me with GEN4 this gives me the total energy today.
After restart, can you please send me the log again, maybe there is something during startup

flopp999, 2025-08-11:
> @flopp999 , i'm not sure where to put 3083?

Where it says address: 59

Change 59 to 3083

flopp999, 2025-08-11:
Can you change to address 63?

61 was for Total Life time solar energy PV1

flopp999, 2025-08-11:
> Yes, I have 2 strings. Could se effect (watt) on both before. So I will disable the growatt.yaml and hopefully get data again.

you meant modbus.yaml I think

Laksepote, 2025-08-11:
> > Yes, I have 2 strings. Could se effect (watt) on both before. So I will disable the growatt.yaml and hopefully get data again.
> 
> you meant modbus.yaml I think

Of course. Heres the log

[home-assistant_solax_modbus_2025-08-11T17-35-50.533Z.log](https://github.com/user-attachments/files/21720216/home-assistant_solax_modbus_2025-08-11T17-35-50.533Z.log)

flopp999, 2025-08-11:
> [image]

this error is not a bug in the integration. All those 3 datas is directly from inverter, so maybe Growatt is doing a mistake

flopp999, 2025-08-11:
> Funny... Now todays solar energy changed from 0 to the value of pv1. However. I know the value of pv2 is not "0"
> 
> [image]

yes, that was funny. Can you check what your second MPPT are showing in shinephone?
What is the total today solar energy in the integration and what do you se in shinephone?

flopp999, 2025-08-11:
> Funny... Now todays solar energy changed from 0 to the value of pv1. However. I know the value of pv2 is not "0"
> 
> [image]

is PV1 today solar energy correct? Same number in shinephone?

Laksepote, 2025-08-11:
Sorry, I don't have shinephone

Laksepote, 2025-08-11:
But the values matches the ones I can read in the display on the inverter. Maybe problem is solved by just making the pv1 and pv2 sensors visible.

Laksepote, 2025-08-11:
Just went back from vacation. Maybe my pv2 is broken..

flopp999, 2025-08-11:
> > > Sorry, I don't have shinephone
> > 
> > can you check on webpage, https://server.growatt.com or https://shineserver.growatt.com/
> 
> But my inverter is offline.

Ah ok, yes it goes to sleep when sun is down, it wakes up tomorrow he sun is up :)

Laksepote, 2025-08-11:
Just checked the History. My pv2 dropped to 0 this afternoon. However. It seems that the total solar power is working now. Very strange, but it changed when i activated the 2 sensors.

flopp999, 2025-08-11:
> Just checked the History. My pv2 dropped to 0 this afternoon. However. It seems that the total solar power is working now. Very strange, but it changed when i activated the 2 sensors.

Strange.
Lets see what happens tomorrow after a few hours of sunshine

fontexD, 2025-08-16:
Having the same issue on sph3600

Tried restart and dig logs, can I enable more debugging in the code? I'll try dig into the issue as well

[image]

But it seems it doesn't throw errors as such just not getting any values

[image]

fontexD, 2025-08-16:
Got it working by enabling pv1 and pv2 today solar > Having the same issue on sph3600
> 
> Tried restart and dig logs, can I enable more debugging in the code? I'll try dig into the issue as well
> 
> [image]
> 
> 
> But it seems it doesn't throw errors as such just not getting any values
> 
> 
> [image]
> 
>

closed 2026-10-03 by github-actions[bot]