wills106/homeassistant-solax-modbus #1508: [FR]: Update the state of the sensors faster when using different scangroup intervals

infradom, 2025-07-30:
### Description

I recently started to use different scan intervals on Solax. The fast changing entities are now polled every 3 seconds while the slower changing (default) entities are polled every 30 seconds in my setup.
I noticed that the GUI status updates come in at the slowest rate for computed sensors (like pv_power_total).

It would be better if these status updates are sent at the fastest rate to the GUI and automations.

I am preparing a PR that resolves this issue/feature request. I just wanted to post this to hear feedback before I proceed further.

### Suggested Solution

Perform a modbus_data_update() call after the value_functions are called.  
Disabled entities should not be computed also. For internal entities, the state need not be updated

rosenrot00, 2025-07-30:
What you have in mind makes perfect sense to me. As you pointed out, calculated values should always reflect the fastest-changing input variable they depend on.

Regarding disabled entities. I guess as you read address blocks it makes sense still to read blocks containing registers of disabled entities but just not to process them, right?

infradom, 2025-08-01:
Implemented

closed 2025-08-01 by infradom

infradom, 2025-08-01:
> Great! Is it included in the next release?

If I remember well, it is already included in the beta release 2025.07.8_B1

rosenrot00, 2025-08-01:
> > Great! Is it included in the next release?
> 
> If I remember well, it is already included in the beta release 2025.07.8_B1

I can confirm that fast polling every 5s does poll sensor values. 

The slow group is now the medium polling interval?

What is the first setting for called something like interval for modbus registers?

infradom, 2025-08-01:
> The slow group is now the medium polling interval?
For Solax, the plugin specifies following default mechanism (see also readTheDocs, developer guide)
```
plugin_instance = solax_plugin(
    plugin_name="SolaX",
    ....
    default_holding_scangroup=SCAN_GROUP_MEDIUM, 
    default_input_scangroup=SCAN_GROUP_AUTO, # SCAN_GROUP_MEDIUM for slow changing units like temperature, kWh, ...
    auto_default_scangroup=SCAN_GROUP_FAST,
    auto_slow_scangroup=SCAN_GROUP_MEDIUM,
)
```
So the default for holding registers is MEDIUM, 
for input registers: AUTO, which means MEDIUM for slow changing parameters like temperature and kWh, otherwise FAST

The real intervals behind these 3 names can be seen by reconfiguring your integration.