wills106/homeassistant-solax-modbus #1511: [Bug]: Sensor House Load shows incorrect negative values in 2025.07.7 (worked correctly in 2025.07.3 and 2025.07.6)

CVKBaca, 2025-07-30:
### Describe the bug

The sensor House Load reports incorrect negative values (e.g. -2286 W) in integration version 2025.07.7. 

[image]
In 2025.07.3 and 2025.07.6, the sensor correctly showed home load (e.g. 708 W).
In 2025.07.4, 2025.07.5, and 2025.07.7, the sensor appears to display PV production / inverter output instead of the real house load.
Sometimes it even shows negative values (e.g. −2 286 W), which is not realistic for household consumption.
This error also appeared in versions 2025.07.4 and 2025.07.5, though version 2025.07.6 seems to restore correct behavior. 
The latest version where House Load worked correctly was 2025.07.3, e.g. showing correct values as expected:
[image]

Expected behavior
House Load should display the actual house consumption, not the PV production from the inverter.

**Actual behavior**

- Shows inverter output instead of house consumption
- Occasionally reports negative values

Other sensors (Grid Export/Import, inverter power, measured power, currents, voltages) appear to be working normally.

### Integration Version

2025.07.7

### Homeassistant core version

2025.7.4

### Inverter brand

SolaX

### Plugin used

plugin_solax.py

### Serial prefix

MU902T

### Inverter firmware versions

DSP v1.38, ARM v1.33

### Connection Method

Waveshare RS485 to RJ45 Ethernet DIN Rail Mount - RS485 TO ETH (B)

### Dongle firmware

Waveshare : V1. 486

infradom, 2025-07-31:
House load is computed with following function:
```

def value_function_house_load(initval, descr, datadict):
    return (
        datadict.get("inverter_power", 0)
        - datadict.get("measured_power", 0)
        + datadict.get("meter_2_measured_power", 0)
    )

```
Can you also show inverter_power, measured_power and meter_2_measured_power at the same time ?

One other questionn: do you use different polling intervals in your configuration?

A log file would help

infradom, 2025-07-31:
Sorry, the answer was in your screenshot: inverter_power seems to be missing ...
A debug logfile would help

infradom, 2025-07-31:
There are no relevant changes in the code history, but the config file does not contain a matching entry for your inverter type MIC |  GEN | X3
That it worked in the past is a bit unclear to me. Asking help from @wills106 

Of course a debug log file (containing a HA restart) would make our life so much easier ...

infradom, 2025-07-31:
@wills106 has just published a bete release 2025.07.8b1 , it might solve your problem, but I am not sure yet.
I would appreciate it if you coud test it.

CVKBaca, 2025-07-31:
Thanks for checking the code history. I realized that the root cause was that the inverter_power entity was disabled/hidden in my setup.

After enabling it, the House Load sensor started reporting the correct home consumption values again in 2025.07.7.

So the issue was related to the recent change that the integration no longer reads disabled entities – which explains why it worked in 2025.07.3/07.6 and failed in 07.4/07.5/07.7.

Everything works fine now. Thanks for the clarification!

[image]

closed 2025-08-01 by infradom