wills106/homeassistant-solax-modbus #1523: [Bug]: 'Solis' - 'RHI-6K-48ES-5G-DC' Inconsistent values

mooky31, 2025-08-05:
### Describe the bug

Hello,

Since I upgraded from 2025.07.3, I have inconsistent values.

[image]
On this image, grid input + solar input + battery output should be around battery input + house load (modulo the inverter consumption, around 200w).
Here they are completely off.

Which additional info could help you solve this ?

### Integration Version

2025.07.07

### Homeassistant core version

2025.7.4

### Inverter brand

Solis

### Plugin used

plugin_solis.py

### Serial prefix

Unknown

### Connection Method

Dongle S2

mooky31, 2025-08-05:
As counters in kWh are right, my feeling is that values that were before gathered at the same time are now gathered at different times. And since those numbers move quickly, if they are not gathered are the same time they are inconsistent.
Do recent changes may have caused this ?
Sorry, I am not skilled enough to find this myself from repository history.

robjwalker, 2025-08-07:
I'm seeing something similar since the last update, and it is because battery output power is showing as input power. So it shows the battery charging when it is actually discharging.

mooky31, 2025-08-07:
Input 1337 + 3517 + 118 = 4972
Output 4718
=> robjwalker theory seems right

infradom, 2025-08-07:
Thanks @robjwalker 

You must enable battery_charge_direction, it is probably disabled (was disabled by default; the new framework does not read registers that are disabled)
I have an open PR that should solve this problem. Once @wills106 has merged it and built a new (beta) release, it

infradom, 2025-08-07:
The battery power computation uses the battery_charge_direction. The initial description of the problem did not mention which sensor was probably wrong. Thanks @robjwalker for pointing out which sensor was wrong.

mooky31, 2025-08-08:
So I reupgraded to 2025.7.7, and activated battery power and battery charge direction.

[image]

931+2600+10 ~= 3340

Thanks @robjwalker and @infradom !

closed 2025-08-08 by infradom