wills106/homeassistant-solax-modbus #1531: [Bug]: MAX_EXPORT concept does not work when multiple inverters and/or master-slave config is present

jprates, 2025-08-09:
### Describe the bug

WHAT IS WRONG:
plugin_solax.py assumes for MAX_EXPORT the maximum output of each inverter type/model.

WHY:
When we have multiple vendor or even just a master/slave configuration on our solar setup this is just not true.

The mid meter (or CT clamps) measure the power going out at grid connection interface, no matter how many inverters you have, and it is that measured value that is going to be transmitted and evaluated by the inverter in parameter "Export Control".

Plus, when you are in master/slave mode, some parameters, as is the case of "Export Control" are to be set at the master inverter, and it is only this inverter that has connection to the mid meter and sends control commands to the slave, as with many other parameters, such as charge/discharge schedule for example, it's only set on master's side.

EXAMPLE SCENARIO:
Let's look at a more complex scenario for the sake of clarity:
You have 2 x 10 kW Gen4 Solax Hybrid working in master/slave config, together producing 14 kW at a given point in time.
You also have a 5 kW SMA inverter working in stand alone, producing 3 kW at the same point in time.
Your consumption is 4 kW at the same point in time.

The SOLAX master will see the current flowing out to the grid, 15 kW of power are going out.
If you change the export parameter to 10 kW, the solax cluster will lower its production to 7 kW, because the meter reading is what commands, the energy flow, no matter how many inverters you have.

Case in point:
If you want to remove the export limit and let all inverters produce freely to the maximum and inject whatever is left from consumption into the grid, the export limit must be 2 x 10 + 5 = 25 kW to accommodate for consumption == 0.
Of course you should take care this value should match the main breaker current limit, but that's another story.

You can NOT limit the value on the max_export parameter to the maximum power generated by the inverter with the MID meter, that's a very specific use case when you have only 1 inverter in your setup.

For my setup and my grid scenario I had to change the  ("H34A10", 15000)  for  ("H34A10", 21000) because I needed to setup for 30A i.e. 20700W, but not all users will be able to understand this is where the bottleneck was.

### Integration Version

2025.07.7

### Homeassistant core version

2025.8.0

### Inverter brand

SolaX Power

### Plugin used

plugin_solax.py

### Serial prefix

H34A10

### Inverter firmware versions

DSP v1.48 ARM v1.45

### Connection Method

SolaX Pocket WiFi

### Dongle firmware

3.012.01

### Additional context

Value 20700 for number.solax10_export_control_user_limit is outside valid range 0 - 15000.

infradom, 2025-08-09:
I understand, maybe we should create a new WRITE_DATA_LOCAL entity by which one can set the desired maximum_export.
I would keep the current mechansm to provide the default value.
Lets think about it a couple of days ...

infradom, 2025-08-10:
Sorry, I had a déjà vu feeling when looking into this: 
I believe we have already implemented what you want (or similar). There is a disabled entity called `config_max_export`. If you enable that entity, you can modify the max_export level. 
It is possible that a restart is needed to adapt the sliders (probably not)

Can you try this?

jprates, 2025-08-10:
Sure.

There was a disabled entity indeed, on the diagnostic section, and it works, it does change the export limit.

I'll close this issue with the recommendation that this entity is documented so that other people don't open more issues like this in the future.

Thanks for taking your time, and care, to look at this!

jprates, 2025-08-10:
disabled `config_max_export `entity exists to configure the upper limit of `export_control_user_limit`.
Recommend to document this disabled entity in project docs.

closed 2025-08-10 by jprates