wills106/homeassistant-solax-modbus #1533: [Bug]: Integration faiiled HASS 2025.8

tiaxter, 2025-08-12:
After upgrading Home Assistant to the version 2025.8 the integration loading failed showing these errors:
```
Logger: homeassistant.setup
Source: setup.py:343
First occurred: 12:47:37 PM (1 occurrence)
Last logged: 12:47:37 PM

Setup failed for custom integration 'solax_modbus': Unable to import component: No module named 'pymodbus.payload'
Traceback (most recent call last):
  File "/nix/store/p2b2wgc2d6b5nymp1fibacf92mq9smw6-homeassistant-2025.8.0/lib/python3.13/site-packages/homeassistant/setup.py", line 343, in _async_setup_component
    component = await integration.async_get_component()
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/nix/store/p2b2wgc2d6b5nymp1fibacf92mq9smw6-homeassistant-2025.8.0/lib/python3.13/site-packages/homeassistant/loader.py", line 1023, in async_get_component
    self._component_future.result()
    ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^^
  File "/nix/store/p2b2wgc2d6b5nymp1fibacf92mq9smw6-homeassistant-2025.8.0/lib/python3.13/site-packages/homeassistant/loader.py", line 1003, in async_get_component
    comp = await self.hass.async_add_import_executor_job(
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
        self._get_component, True
        ^^^^^^^^^^^^^^^^^^^^^^^^^
    )
    ^
  File "/nix/store/cr7c6cf8y7sbqwq9bj7lrm977x7c4mv9-python3-3.13.5/lib/python3.13/concurrent/futures/thread.py", line 59, in run
    result = self.fn(*self.args, **self.kwargs)
  File "/nix/store/p2b2wgc2d6b5nymp1fibacf92mq9smw6-homeassistant-2025.8.0/lib/python3.13/site-packages/homeassistant/loader.py", line 1063, in _get_component
    ComponentProtocol, importlib.import_module(self.pkg_path)
                       ~~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^
  File "/nix/store/p2b2wgc2d6b5nymp1fibacf92mq9smw6-homeassistant-2025.8.0/lib/python3.13/site-packages/homeassistant/util/loop.py", line 201, in protected_loop_func
    return func(*args, **kwargs)
  File "/nix/store/cr7c6cf8y7sbqwq9bj7lrm977x7c4mv9-python3-3.13.5/lib/python3.13/importlib/__init__.py", line 88, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
           ~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "<frozen importlib._bootstrap>", line 1387, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1360, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1331, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 935, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 1026, in exec_module
  File "<frozen importlib._bootstrap>", line 488, in _call_with_frames_removed
  File "/var/lib/hass/custom_components/solax_modbus/__init__.py", line 59, in <module>
    from .sensor import SolaXModbusSensor, is_entity_enabled
  File "/var/lib/hass/custom_components/solax_modbus/sensor.py", line 12, in <module>
    from .const import ATTR_MANUFACTURER, DOMAIN, SLEEPMODE_NONE, SLEEPMODE_ZERO
  File "/var/lib/hass/custom_components/solax_modbus/const.py", line 15, in <module>
    from pymodbus.payload import Endian
ModuleNotFoundError: No module named 'pymodbus.payload'
```

I noticed that in HASS 2025.8 the `pymodbus` version has bumped to `3.11` and in `pymodbus@3.11` the `pymodbus.payload` module has been removed. Another thing is that `pymodbus.constants` has no more `Endian` variable.

Here some reference:
- [HASS 2025.8 Bump Version](https://github.com/home-assistant/core/pull/150419)

tiaxter, 2025-08-20:
I updated the components to its latest version but now the error is still the same:
```
Logger: homeassistant.setup
Source: setup.py:343
First occurred: 4:42:14 PM (1 occurrence)
Last logged: 4:42:14 PM

Setup failed for custom integration 'solax_modbus': Unable to import component: No module named 'pymodbus.payload'
Traceback (most recent call last):
  File "/var/lib/hass/custom_components/solax_modbus/const.py", line 16, in <module>
    from pymodbus.constants import Endian  # for pymodbus < 3.0
    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
ImportError: cannot import name 'Endian' from 'pymodbus.constants' (/nix/store/jvvdyyfbrzflrx2ad39n7a8r43b98bkp-python3.13-pymodbus-3.11.1/lib/python3.13/site-packages/pymodbus/constants.py)

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/nix/store/068slax0cvh18mncnxs0q4g8xhk1zczv-homeassistant-2025.8.2/lib/python3.13/site-packages/homeassistant/setup.py", line 343, in _async_setup_component
    component = await integration.async_get_component()
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/nix/store/068slax0cvh18mncnxs0q4g8xhk1zczv-homeassistant-2025.8.2/lib/python3.13/site-packages/homeassistant/loader.py", line 1023, in async_get_component
    self._component_future.result()
    ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^^
  File "/nix/store/068slax0cvh18mncnxs0q4g8xhk1zczv-homeassistant-2025.8.2/lib/python3.13/site-packages/homeassistant/loader.py", line 1003, in async_get_component
    comp = await self.hass.async_add_import_executor_job(
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
        self._get_component, True
        ^^^^^^^^^^^^^^^^^^^^^^^^^
    )
    ^
  File "/nix/store/cr7c6cf8y7sbqwq9bj7lrm977x7c4mv9-python3-3.13.5/lib/python3.13/concurrent/futures/thread.py", line 59, in run
    result = self.fn(*self.args, **self.kwargs)
  File "/nix/store/068slax0cvh18mncnxs0q4g8xhk1zczv-homeassistant-2025.8.2/lib/python3.13/site-packages/homeassistant/loader.py", line 1063, in _get_component
    ComponentProtocol, importlib.import_module(self.pkg_path)
                       ~~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^
  File "/nix/store/068slax0cvh18mncnxs0q4g8xhk1zczv-homeassistant-2025.8.2/lib/python3.13/site-packages/homeassistant/util/loop.py", line 201, in protected_loop_func
    return func(*args, **kwargs)
  File "/nix/store/cr7c6cf8y7sbqwq9bj7lrm977x7c4mv9-python3-3.13.5/lib/python3.13/importlib/__init__.py", line 88, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
           ~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "<frozen importlib._bootstrap>", line 1387, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1360, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1331, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 935, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 1026, in exec_module
  File "<frozen importlib._bootstrap>", line 488, in _call_with_frames_removed
  File "/var/lib/hass/custom_components/solax_modbus/__init__.py", line 59, in <module>
    from .sensor import SolaXModbusSensor
  File "/var/lib/hass/custom_components/solax_modbus/sensor.py", line 12, in <module>
    from .const import ATTR_MANUFACTURER, DOMAIN, SLEEPMODE_NONE, SLEEPMODE_ZERO
  File "/var/lib/hass/custom_components/solax_modbus/const.py", line 18, in <module>
    from pymodbus.payload import Endian    # for pymodbus >= 3.0
    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
ModuleNotFoundError: No module named 'pymodbus.payload' 
```

rosenrot00, 2025-08-20:
@tiaxter can you edit your const.py file to see if this fix works? I didn't check the entire code, but if you can change the file, let's see if it crashes somewhere else or if this fixes it.

Replace this:
```
try:
    from pymodbus.constants import Endian  # for pymodbus < 3.0
except ImportError:
    from pymodbus.payload import Endian    # for pymodbus >= 3.0
```
With this:
```
from enum import Enum

try:
    from pymodbus.constants import Endian
except Exception:
    try:
        from pymodbus.payload import Endian
    except Exception:
        class Endian(Enum):
            BIG = "big"
            LITTLE = "little"
```

tiaxter, 2025-08-20:
@rosenrot00 thanks for the suggestion. I made a PR fixing the problem starting from your idea

infradom, 2025-09-01:
Now that most major changes are hopefully behind us, it may be time to convert our code from `payload` to `DATATYPE/convert_to/from_registers` method. The change is not so straightforward as we might hope.
Just a question to avoid that we do double work: who will take the lead in this conversion ?

rosenrot00, 2025-09-01:
@infradom please feel free to take the lead as I'm rather busy this week. I've already started something offline, but it isn't working yet. I get the readings correctly, but there is a conversation (LSM/MSB) error, and other things to do. I tried to adapt it to the latest codebase, but likely there might be some errors. Take what you see as useful. From what I have done, you can understand what I have in mind (do compatibility things for pymodbus in a single file and make it future proof so that in the best case we can at some point remove the compatibility file and it still works). 

https://github.com/rosenrot00/homeassistant-solax-modbus/tree/universal-pymodbus

The __init__.py is still based on the old codebase, I haven't found time to adapt it.

infradom, 2025-09-01:
A quick question (if you have time): the new pymodbus api does not pass byteorder, only wordorder.
Your compat layer still seems to pass both to the new API which looks strange to me, as according to the documentation and source code, this byteorder argument is not implemented.

Al our plugins seem to use BIG for byteorder, the difference is only for wordorder, so we will not miss byteorder. In my opinion, we should remove it from the new compat interface.

infradom, 2025-09-01:
I take my time... Merged your code into the latest line, and made some minor modifications. I try to stay close to the new pymodbus API, so that all the stuff related with the old model remains in pymodbus-compat

I am ready to start testing, I expect a long debugging time. I cannot test yet; waiting until my inverter has finished the remotecontrol discharge phase

infradom, 2025-09-01:
There is a dedicated branch on my github for this

rosenrot00, 2025-09-02:
I completely understand why it looks like a promising approach to use client. Very often just to discuss with someone else helps a lot to get a different perspective.

Did you try this to probe if the functions exist?

```
import pymodbus

_cfr = getattr(pymodbus, "convert_from_registers", None)
_ctr = getattr(pymodbus, "convert_to_registers", None)
```

wills106, 2025-09-02:
Just released [2025.09.2](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/2025.09.2) with the temporary fix from #1551

infradom, 2025-09-02:
yes, thanks, I hope this does not cause problems (see remark from @fight1wer1) but maybe he applied the patch on a different software version

rosenrot00, 2025-09-02:
Thanks for that, I just scrolled through.

Some further points (with which do what you think is right).

We could just import everything and use it, no need to set datatypes globally.
`from .pymodbus_compat import DataType, convert_to_registers, convert_from_registers, pymodbus_version_info`

If the `getattr(` works, there isn't a strong need for the client anymore, right? We wouldn't lose much beauty.

infradom, 2025-09-02:
I agree completely. Thanks. 
I will now need to help my neighbour with an electricity leak problem, dont know how long this will take, I might be away for hours if it is hard to find.

infradom, 2025-09-02:
Had a quick look: I dont think your DataType declaration is compatible with the new pymodbus DATATYPE
Why do you use a different type? Once the old installatons are gone, we should not have to touch much code in __init__.py
I probably overlooked something with this remark .. .so  
Your _oldEndian declaration is probably not the correct definition. There was a definition wiht "<" and ">"

Sorry for this quick reaction, need to continue my electricity work ...
Found the neighbors earth-leak but need to spend some more time on the repair now

rosenrot00, 2025-09-02:
> “Failed to perform the action button/press. ModbusClientMixin.write_registers() got an unexpected keyword argument 'slave'”

This is also a pymodbus issue, which wasn't resolved yet but should be with what we are working on currently. To mitigate your issue, change your local pymodbus version to one that still supports "slave".

markjsmith, 2025-09-02:
Hi Rosenrot, is that a suggestion for me? If so am I looking on the modbus itself?

infradom, 2025-09-02:
@rosenrot00  This seems to work (without passing the client object):

```

# ---------------- Try to access new helpers (≥ 3.9) ------------------
_convert_to   = None
_convert_from = None
if _USE_NEW_API and pymodbus and pymodbus.client and pymodbus.client.mixin and pymodbus.client.mixin.ModbusClientMixin :
    _convert_to = getattr(pymodbus.client.mixin.ModbusClientMixin, "convert_to_registers", None)
    _convert_from = getattr(pymodbus.client.mixin.ModbusClientMixin, "convert_from_registers", None)
    if not callable(_convert_to) or not callable(_convert_from):
        _convert_to = _convert_from = None

```

rosenrot00, 2025-09-02:
I have seen your remarks but as you have a good looking and working version I think we are fine. We can still tweak the compat file if needed.

infradom, 2025-09-02:
Shall we ask @wills106 for a beta, or do you propose a normal release?

rosenrot00, 2025-09-02:
Let‘s be brave and do a normal release.

If @wills106 is around and can push another one in case we missed something we have it working before the vast majority of people even realise there is a new release.

infradom, 2025-09-02:
PR #1587 created; Copilot review requested

infradom, 2025-09-02:
plugin_solax cannot read the serial number:  seems to interpret is as bits

``` 
2025-09-02 18:40:16.980 INFO (MainThread) [custom_components.solax_modbus.plugin_solax] Read SolaX 0x0 serial number: [False, False, False, True, False, False, True, False, True, True, False, False, True, True, False, False, False, False, True, False, True, True, False, False, False, False, True, False, True, False, True, False, False, False, False, False, True, True, False, False, False, False, False, True, True, True, False, False, False, False, False, True, False, False, True, False, True, True, True, False, True, True, False, False, True, False, False, False, True, True, False, False, True, False, False, True, True, True, False, False, False, False, False, False, True, True, False, False, False, False, False, False, True, True, False, False, True, False, False, False, True, True, False, False, True, True, False, False, True, True, False, False]
```

rosenrot00, 2025-09-02:
I think the issue is in this path:
`# ---------------- New helper API path (≥ 3.9.0) ----------------------`
It seems the datatype is not correctly provided in enum and thus it isn't interpreted correctly.

wills106, 2025-09-03:
Seems to be fine for me so far.

infradom, 2025-09-03:
Just tested with plugin_solax: seems to work now. thanks
Other questions:

- I looked at the pymodbus source code and only found the convert_ functions and DATATYPE in the mixin.  Did you find it in other locations?

- why dont we use DATATYPE instead of DataType and the _coerce_dt function (looks like an overly complicated piece of code)?  Sorry for this remark ;-)

wills106, 2025-09-03:
I didn't realise you updated all the other plugins locally. I have updated them here. Hopefully I have done them the same way you have.

wills106, 2025-09-03:
Just released [2025.09.4](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/2025.09.4) with the fixes for newer pyModbus

rosenrot00, 2025-09-03:
@markjsmith could you give the latest release a try?

markjsmith, 2025-09-03:
@rosenrot00 still the same issue, got the log for you:

<logger:%20homeassistant.components.websocket_api.http.connection%0ASource%3A%20components/websocket_api/commands.py:264%0Aintegration:%20Home%20Assistant%20WebSocket%20API%20(documentation,%20issues)%0AFirst%20occurred:%2010:21:24%20(1%20occurrence)%0ALast%20logged:%2010:21:24%0A%0A%5B140211992113248%5D%20Unexpected%20exception%0ATraceback%20(most%20recent%20call%20last):%0A%20%20File%20%22/usr/src/homeassistant/homeassistant/components/websocket_api/commands.py%22,%20line%20264,%20in%20handle_call_service%0A%20%20%20%20response%20=%20await%20hass.services.async_call(%0A%20%20%20%20%20%20%20%20%20%20%20%20%20%20%20%5E%5E%5E%5E%5E%5E%5E%5E%5E%5E%5E%5E%5E%5E%5E%5E%5E%5E%5E%5E%5E%5E%5E%5E%5E%5E%5E%5E%5E%5E%5E%0A%20%20%20%20...%3C7%20lines%3E...%0A%20%20%20%20)%0A%20%20%20%20%5E%0A%20%20File%20%22/usr/src/homeassistant/homeassistant/core.py%22,%20line%202835,%20in%20async_call%0A%20%20%20%20response_data%20=%20await%20coro%0A%20%20%20%20%20%20%20%20%20%20%20%20%20%20%20%20%20%20%20%20%5E%5E%5E%5E%5E%5E%5E%5E%5E%5E%0A%20%20File%20%22/usr/src/homeassistant/homeassistant/core.py%22,%20line%202878,%20in%20_execute_service%0A%20%20%20%20return%20await%20target(service_call)%0A%20%20%20%20%20%20%20%20%20%20%20%5E%5E%5E%5E%5E%5E%5E%5E%5E%5E%5E%5E%5E%5E%5E%5E%5E%5E%5E%5E%5E%5E%5E%5E%5E%5E%0A%20%20File%20%22/usr/src/homeassistant/homeassistant/helpers/service.py%22,%20line%20850,%20in%20entity_service_call%0A%20%20%20%20single_response%20=%20await%20_handle_entity_call(%0A%20%20%20%20%20%20%20%20%20%20%20%20%20%20%20%20%20%20%20%20%20%20%5E%5E%5E%5E%5E%5E%5E%5E%5E%5E%5E%5E%5E%5E%5E%5E%5E%5E%5E%5E%5E%5E%5E%5E%5E%5E%0A%20%20%20%20%20%20%20%20hass,%20entity,%20func,%20data,%20call.context%0A%20%20%20%20%20%20%20%20%5E%5E%5E%5E%5E%5E%5E%5E%5E%5E%5E%5E%5E%5E%5E%5E%5E%5E%5E%5E%5E%5E%5E%5E%5E%5E%5E%5E%5E%5E%5E%5E%5E%5E%5E%5E%5E%5E%0A%20%20%20%20)%0A%20%20%20%20%5E%0A%20%20File%20%22/usr/src/homeassistant/homeassistant/helpers/service.py%22,%20line%20922,%20in%20_handle_entity_call%0A%20%20%20%20result%20=%20await%20task%0A%20%20%20%20%20%20%20%20%20%20%20%20%20%5E%5E%5E%5E%5E%5E%5E%5E%5E%5E%0A%20%20File%20%22/usr/src/homeassistant/homeassistant/components/select/__init__.py%22,%20line%20188,%20in%20async_handle_select_option%0A%20%20%20%20await%20self.async_select_option(option)%0A%20%20File%20%22/config/custom_components/solax_modbus/select.py%22,%20line%20118,%20in%20async_select_option%0A%20%20%20%20await%20self._hub.async_write_register(unit=self._modbus_addr,%20address=self._register,%20payload=payload)%0A%20%20File%20%22/config/custom_components/solax_modbus/__init__.py%22,%20line%20729,%20in%20async_write_register%0A%20%20%20%20return%20await%20self.async_lowlevel_write_register(unit,%20address,%20payload)%0A%20%20%20%20%20%20%20%20%20%20%20%5E%5E%5E%5E%5E%5E%5E%5E%5E%5E%5E%5E%5E%5E%5E%5E%5E%5E%5E%5E%5E%5E%5E%5E%5E%5E%5E%5E%5E%5E%5E%5E%5E%5E%5E%5E%5E%5E%5E%5E%5E%5E%5E%5E%5E%5E%5E%5E%5E%5E%5E%5E%5E%5E%5E%5E%5E%5E%5E%5E%5E%5E%5E%5E%0A%20%20File%20%22/config/custom_components/solax_modbus/__init__.py%22,%20line%20722,%20in%20async_lowlevel_write_register%0A%20%20%20%20resp%20=%20await%20self._client.write_register(address,%20regs%5B0%5D,%20**kwargs)%0A%20%20%20%20%20%20%20%20%20%20%20%20%20%20%20%20%20~~~~~~~~~~~~~~~~~~~~~~~~~~~%5E%5E%5E%5E%5E%5E%5E%5E%5E%5E%5E%5E%5E%5E%5E%5E%5E%5E%5E%5E%5E%5E%5E%5E%5E%5E%5E%5E%0ATypeError:%20ModbusClientMixin.write_register()%20got%20an%20unexpected%20keyword%20argument%20'slave'%0A>

rosenrot00, 2025-09-03:
> Just realised I missed removing some of the Endian's, could you push it as a PR please?

https://github.com/wills106/homeassistant-solax-modbus/pull/1595

I hope I didn't overlook anything - it is just getting better :)

rosenrot00, 2025-09-03:
@markjsmith any chance that you could provide us an entire debug log (with HA restart) or check in your environment which pymodbus is installed? Slave was introduced a long time ago and I‘m wondering why it goves errors on your end. Either you have an old version or a very new one installed.

rosenrot00, 2025-09-03:
@tiaxter and @frkca could you test as well as you seem to have exotic pymodbus installations.

rosenrot00, 2025-09-03:
@markjsmith thank you that is really helpful. Your logs will help a lot. I will also try to update my system to 3.11.1 later.

Sorry for the hassle, there are many different pymodbus versions around and we try to catch them even down to 3.7.4. However, docker containers seem not to use 3.11.1 yet.

markjsmith, 2025-09-03:
Hi again @rosenrot00 . Please find the log attached, a bit long I’m afraid!

[home-assistant_solax_modbus_2025-09-03T12-54-05.197Z.log](https://github.com/user-attachments/files/22119129/home-assistant_solax_modbus_2025-09-03T12-54-05.197Z.log)

rosenrot00, 2025-09-03:
Can you try to catch that in a log file? I just want to make sure that the error didn‘t change. Does it happen on any change of setting or only a specific one?

infradom, 2025-09-03:
> I introduced a fast-path. The fast-path should exactly do what you intend.
> 
> https://github.com/rosenrot00/homeassistant-solax-modbus/tree/universal-pymodbus
Super !
I am back now and would like to test your fast-path , but I don't see relevant changes in the link you sent

markjsmith, 2025-09-03:
Brill  😊

rosenrot00, 2025-09-04:
@infradom I just checked your draft PR regarding the compat layer and it looks cleaner than before. I would say that makes at even easier to maintain. Well done.

wills106, 2025-09-04:
Just released [2025.09.6](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/2025.09.6)

rosenrot00, 2025-09-04:
Fingers crossed, we caught all these pymodbus issues now.

@wills106 payload.py should be deprecated now :)

rosenrot00, 2025-09-04:
@infradom actually now that I'm thinking. Why do we still need the payload.py path? BinaryPayloadDecoder and co go back to before version 2.3.0 - released in 2018/19. I would say it is fair to remove that path and create an error message. I guess running such an instance has other issues than pymodbus.

infradom, 2025-09-04:
yes indeed, I have had that thought many times, but I felt I had too many open issues to introduce yet another potential problem ;-)

Indeed, it might become time to get rid of it.

rosenrot00, 2025-09-04:
I’d say it’s about time you finally create a PR for that.

infradom, 2025-09-04:
> After that, the bad register auto-detection seems to need our attention. It looks like it doesn't analyse all the blocks but just the first one and thus not detecting bad registers in all the blocks.

I must admin that in my limited tests I chose registers from the first holding block if I remember well.

infradom, 2025-09-04:
> I’d say it’s about time you finally create a PR for that.

I still have one outstanding sofar problem #1503 caused  by may "don't read disabled registers anymore" change. 
I would give the bad-register autodetection priority after a new investigation in the sofar problem #1503 

Then the payload ....

infradom, 2025-09-04:
Further analysis on issue #1503 shows that it is related to a block read error (not the first block). So we might be chasing the same issue ...
I move my priority to debugging the bad-register autodetection code therefore

infradom, 2025-09-04:
Update after faulty debug code/observations: 
Using second holding block 0x104-0x161 in these tests:

- I invalidated holding register 0x120 in block 0x104-0x161 , seems to split ok:
- I invalidated holding register 0x123 in block 0x104-0x161 , seems to split ok:
- I invalidated holding register 0x106 in block 0x104-0x161 , seems to have shortened the block correctly: ok
- I invalidated holding register 0x160 in block 0x104-0x161 , seems to have shortened the block correctly: ok

rosenrot00, 2025-09-04:
Hmmm, for him it doesn't seem to work. If I look at the log bisect doesn't find a faulty register.

https://github.com/wills106/homeassistant-solax-modbus/issues/1543#issuecomment-3245232591

infradom, 2025-09-04:
I think we can close this thread since the discussion is off-topic  now. We can continue the probing discussion in a more appropriate thread