wills106/homeassistant-solax-modbus #1540: [Bug]: 2025.7.7 stopped working with X1 G2

medu80, 2025-08-16:
### Describe the bug

After updating to version 2025.7.7, the integration stops working. All I see are unknown and 0 values. Downgrading to 2025.7.6 resolves the issue. I also tested version 2025.07.8b1.

Additionally, I noticed that the firmware and ARM versions disappear.

There are no indications in the logs that point to the source of the problem or what might be causing it.

### Integration Version

2025.7.7

### Homeassistant core version

2025.8.2

### Inverter brand

Solax Power

### Plugin used

plugin_solax.py

### Serial prefix

XB3350

### Inverter firmware versions

DSP v2.235 ARM v1.152

### Connection Method

Solax Pocket WiFi 3.0

### Dongle firmware

3.007.02

### Detailed Error Log

2025-08-16 13:56:33.251 WARNING (MainThread) [custom_components.solax_modbus.plugin_solax] SolaX: reading serial number from address 0x0 failed; other address may succeed

[image]

[image]

[image]

medu80, 2025-08-21:
[solax_log_hass.txt](https://github.com/user-attachments/files/21918655/solax_log_hass.txt)

infradom, 2025-08-26:
Thanks, @wills106 has released version 2025.8.4 . It solves a couple of problems, but I dont think the mini issue will be solved. It does contain better logging for multi-inverter setups, so that It is easier to see which inverter logs which message.
So I would appreciate it greatly if you could test that version and regenerate the debug logs (including ha restart)
Sorry for the overhead, but there is no way for us to test ourselves, we do not own most inverters.

rosenrot00, 2025-09-01:
@medu80 could you give the latest release a try and post the debug output in any case?

medu80, 2025-09-02:
Hi, the good news is that with the latest version it started working again. I’m not sure which version actually fixed it. I used the X1 device for about six months until I got three-phase power. I no longer have an X1, it was replaced with an X3 yesterday.

closed 2025-09-02 by medu80