wills106/homeassistant-solax-modbus #1542: [Bug]: Solax X1 Mini Inverter all Entities unknown

edli83, 2025-08-17:
### Describe the bug

I have 3 X1 Mini Gen 3 Inverters and 1 X3 Hyb Gen 4 running.
The X3 is connected via LAN Interface and runs very good.
But unfortunately I noticed, that my X1 have trouble.
They are connected via Waveshare RS485 to Eth.

Now I installed earlier Integrations, to test.
With the Version 2025.07.2 they came back and work again.

Could you imagine what happens and hopefully help.

Thank you very much

### Integration Version

2025.07.7

### Homeassistant core version

2025.8.2

### Inverter brand

Solax X1 Mini Gen3

### Plugin used

plugin_solax.py

### Serial prefix

XM3302

### Inverter firmware versions

DSP v2.235 ARM v1.152

### Connection Method

Solax Pocket Wifi 2 and Waveshare RS485 to Eth

### Dongle firmware

Solax Wifi2.0  Dongle  with 3.006.04

### Detailed Error Log

Dieser Fehler stammt von einer benutzerdefinierten Integration

Logger: custom_components.solax_modbus
Quelle: custom_components/solax_modbus/__init__.py:587
Integration: SolaX Inverter Modbus (Dokumentation, Probleme)
Erstmals aufgetreten: 14:24:13 (68 Vorkommnisse)
Zuletzt protokolliert: 15:14:28

Error: device: 70 address: 1536 -> Modbus Error: [Input/Output] No response received after 6 retries, continue with next request
Error: device: 70 address: 1536 -> Modbus Error: [Input/Output] ERROR: No response received of the last requests (default: retries+3), CLOSING CONNECTION.

infradom, 2025-08-17:
I think we will need the complete debug log (including a HA startup)

edli83, 2025-08-17:
Hopefully that is what you need, I cleard the Log then restart Home Assistant and Downlod the Log.

[32m2025-08-17 19:52:00.319 INFO (MainThread) [homeassistant.components.automation.delta_working_mode_laden] Delta Working Mode Laden: Running automation actions[0m
[32m2025-08-17 19:52:00.319 INFO (MainThread) [homeassistant.components.automation.delta_working_mode_laden] Delta Working Mode Laden: Choose at step 1: choice 1: Running automation actions[0m
[32m2025-08-17 19:52:00.319 INFO (MainThread) [homeassistant.components.automation.delta_working_mode_laden] Delta Working Mode Laden: Choose at step 1: choice 1: Executing step call service[0m
[31m2025-08-17 19:52:00.320 ERROR (MainThread) [homeassistant.components.automation.delta_working_mode_laden] Delta Working Mode Laden: Choose at step 1: choice 1: Error executing script. Invalid data for call_service at pos 1: value must be at least 0 @ data['value'][1][0m
[31m2025-08-17 19:52:00.320 ERROR (MainThread) [homeassistant.components.automation.delta_working_mode_laden] Delta Working Mode Laden: Error executing script. Invalid data for choose at pos 1: value must be at least 0 @ data['value'][1][0m
[31m2025-08-17 19:52:00.320 ERROR (MainThread) [homeassistant.components.automation.delta_working_mode_laden] Error while executing automation automation.delta_working_mode_laden: value must be at least 0 @ data['value'][1][0m
[32m2025-08-17 19:52:00.438 INFO (MainThread) [homeassistant.components.automation.automation_delta_phasen_schalten] Delta Phasen Schalten: Running automation actions[0m
[32m2025-08-17 19:52:00.439 INFO (MainThread) [homeassistant.components.automation.automation_delta_phasen_schalten] Delta Phasen Schalten: Choose at step 1: choice 1: Running automation actions[0m
[32m2025-08-17 19:52:01.745 INFO (MainThread) [custom_components.solax_modbus] Inverter is not connected, trying to connect[0m
[32m2025-08-17 19:52:02.438 INFO (MainThread) [homeassistant.components.automation.automation_delta_phasen_schalten] Delta Phasen Schalten: Running automation actions[0m
[32m2025-08-17 19:52:02.438 INFO (MainThread) [homeassistant.components.automation.automation_delta_phasen_schalten] Delta Phasen Schalten: Choose at step 1: choice 1: Running automation actions[0m
[32m2025-08-17 19:52:02.903 INFO (MainThread) [custom_components.solax_modbus.plugin_solax_ev_charger] SolaX_Charger: trying to determine inverter type[0m
[32m2025-08-17 19:52:02.904 INFO (MainThread) [custom_components.solax_modbus] Inverter is not connected, trying to connect[0m
[31m2025-08-17 19:52:04.030 ERROR (MainThread) [custom_components.solax_modbus] Error: device: 70 address: 1536 -> Modbus Error: [Connection] Not connected[AsyncModbusTcpClient 10.21.6.26:502][0m
[33m2025-08-17 19:52:04.031 WARNING (MainThread) [custom_components.solax_modbus.plugin_solax_ev_charger] SolaX_Charger: attempt to read serialnumber failed at 0x600
Traceback (most recent call last):
  File "/config/custom_components/solax_modbus/plugin_solax_ev_charger.py", line 52, in async_read_serialnr
    if not inverter_data.isError():
           ^^^^^^^^^^^^^^^^^^^^^
AttributeError: 'NoneType' object has no attribute 'isError'[0m
[33m2025-08-17 19:52:04.031 WARNING (MainThread) [custom_components.solax_modbus.plugin_solax_ev_charger] SolaX_Charger: reading serial number from address 0x600 failed; other address may succeed[0m
[32m2025-08-17 19:52:04.031 INFO (MainThread) [custom_components.solax_modbus.plugin_solax_ev_charger] Read SolaX_Charger 0x600 serial number before potential swap: None[0m
[31m2025-08-17 19:52:04.031 ERROR (MainThread) [custom_components.solax_modbus.plugin_solax_ev_charger] SolaX_Charger: cannot find serial number for EV Charger[0m
[31m2025-08-17 19:52:04.031 ERROR (MainThread) [custom_components.solax_modbus.plugin_solax_ev_charger] unrecognized inverter type - serial number : unknown[0m
[32m2025-08-17 19:52:04.031 INFO (MainThread) [custom_components.solax_modbus] next inverter check in 10sec[0m
[32m2025-08-17 19:52:04.439 INFO (MainThread) [homeassistant.components.automation.automation_delta_phasen_schalten] Delta Phasen Schalten: Running automation actions[0m
[32m2025-08-17 19:52:04.439 INFO (MainThread) [homeassistant.components.automation.automation_delta_phasen_schalten] Delta Phasen Schalten: Choose at step 1: choice 1: Running automation actions[0m
[32m2025-08-17 19:52:06.438 INFO (MainThread) [homeassistant.components.automation.automation_delta_phasen_schalten] Delta Phasen Schalten: Running automation actions[0m
[32m2025-08-17 19:52:06.438 INFO (MainThread) [homeassistant.components.automation.automation_delta_phasen_schalten] Delta Phasen Schalten: Choose at step 1: choice 1: Running automation actions[0m
[32m2025-08-17 19:52:08.439 INFO (MainThread) [homeassistant.components.automation.automation_delta_phasen_schalten] Delta Phasen Schalten: Running automation actions[0m
[32m2025-08-17 19:52:08.439 INFO (MainThread) [homeassistant.components.automation.automation_delta_phasen_schalten] Delta Phasen Schalten: Choose at step 1: choice 1: Running automation actions[0m
[32m2025-08-17 19:52:10.320 INFO (MainThread) [homeassistant.components.automation.delta_working_mode_laden] Delta Working Mode Laden: Running automation actions[0m
[... cut ...]
[32m2025-08-17 19:53:34.358 INFO (MainThread) [custom_components.solax_modbus] Inverter is not connected, trying to connect[0m
[32m2025-08-17 19:53:34.425 INFO (MainThread) [homeassistant.components.automation.automation_delta_phasen_schalten] Delta Phasen Schalten: Running automation actions[0m
[32m2025-08-17 19:53:34.425 INFO (MainThread) [homeassistant.components.automation.automation_delta_phasen_schalten] Delta Phasen Schalten: Choose at step 1: choice 1: Running automation actions[0m

edli83, 2025-08-17:
sorry, I see too late, that I could have upload also a file - should I?

infradom, 2025-08-17:
On first sight, seems like the error occurs very early in the startup. 
Maybe we have solved this already in the beta release 2025.7.8b1 that we have released.
We have enhanced the error handling (and if we are lucky also the root cause of your problem), But if you could test with that version, and if the problem remains, send the startup log with this beta version

We will hopefully soon release yet another bugfix beta , but the current one should already solve most issues

edli83, 2025-08-17:
I tried the beta release and unfortunatel no change.
How do I get the requested startup log?

infradom, 2025-08-17:
[image]
(activate debug log right upper corner), then restart HA, wait until all services are started and then deactivate the debug log, you will be able to download the log

edli83, 2025-08-17:
[image]

The first is the X3,
then follows the 3 X1,
after this the Solax Wallbox, but currently here I have no values.

infradom, 2025-08-17:
The X1-Mini's have 26 entities, do they show any values ?
The Wallbox serial number is not recognized, this is a different issue. This is what the error mesages refer to.
Are you sure the last logfile is made with 2025.7.8b1 ? (we should output the version in the logfile for verification)

edli83, 2025-08-17:
The manifest.json shows "version": "2025.07.7.1"
So I do the Update, start again, but manifest stays on the old version - I have no idea why !?!

[image]
That´s the screenshot oh the entities of SolaX_02.
Means no values or even unknown ?!?

edli83, 2025-08-17:
Nevertheless, today the X1 should be down due to darkness of night.

edli83, 2025-08-17:
Yes I did install through HACS and choose the beta version.

Wallbox works again after a reboot of itm, means switch off power and on again.

Hopefully we can go on tomorrow
Have a nice evening

edli83, 2025-08-26:
Hi,
thank´s for your effort.
As you expected unfortunately also after update to 2025.8.4 the X1 Mini still have no values. 
See debug Log as attachmant

[home-assistant_solax_modbus_2025-08-26T11-16-22.350Z.log](https://github.com/user-attachments/files/21986760/home-assistant_solax_modbus_2025-08-26T11-16-22.350Z.log)

Hope you will have success with the new Log.

infradom, 2025-08-26:
Thanks for the log, it is now much clearer which inverter sends which message.
The mini's seem to be able to respond to the initial serial number request, the system correctly computes the register blocks to be read, but then for some reason, the polling is not started or blocks at the first modbus request (contradictory observations)

Sorry to say, but maybe we have to issue yet another release with some more extra debug lines. I will first do some more thinking to understand the contradictory observations ...

nromagna, 2025-08-26:
Hi everyone, especially infradom. Same problem after the latest updates, it no longer works. Here's my debug log.

[home-assistant_solax_modbus_2025-08-26T14-55-00.471Z.log](https://github.com/user-attachments/files/21990301/home-assistant_solax_modbus_2025-08-26T14-55-00.471Z.log)

infradom, 2025-08-26:
> Hi everyone, especially infradom. Same problem after the latest updates, it no longer works. Here's my debug log.
> 
> [home-assistant_solax_modbus_2025-08-26T14-55-00.471Z.log](https://github.com/user-attachments/files/21990301/home-assistant_solax_modbus_2025-08-26T14-55-00.471Z.log)
> 
> While with 2025.07.2 everything works.
> 
> X1-Hybrid da SolaX Power Firmware: DSP v1.48 ARM v1.44 Hardware: Gen4

This is not related to the X1 mini  .  This does not belong to this bug report.
The debug log shows that you are certainly not testing with the post recent integration version.  The version this log belongs to is much older than the most recent one. Please report errors based on the most recent version and do not misuse this Mini inverter thread, create a new bug report if you can reproduce this error with  integration version 2025.8.3 or 2025.8.4. A bug report should also mention the HA core version , ...

infradom, 2025-08-28:
> [home-assistant_solax_modbus_2025-08-26T11-16-22.350Z.log](https://github.com/user-attachments/files/21986760/home-assistant_solax_modbus_2025-08-26T11-16-22.350Z.log)
> 
> Hope you will have success with the new Log.

The new latest log only shows the block computing phase for your 4 inverters. I cannot find any read operations; did you wait long enough before generating the log. Please wait long enough after a HA restart, so that we are sure that the system had sufficient time to perform a few polling cycles. This log seems too short and ends only a couple of seconds after the block group computation was finished.

edli83, 2025-08-28:
here is a new Log, waiting more than 10 Minutes to download it..
Hope this will help.

[home-assistant_solax_modbus_2025-08-28T09-24-35.844Z.log](https://github.com/user-attachments/files/22021793/home-assistant_solax_modbus_2025-08-28T09-24-35.844Z.log)

infradom, 2025-08-28:
seems to be something different... I overlooked something in the logs 
There is an attempt to read registers but this fails in block 0x352 and analysis for another user showed that 0x39a is the cause. In my more recent commits, I have (tried) to handle this error (newblock and ignore_readerror), but I have not received feedback as far as I can remember.
I find it a bit strange that you do not test with the most recent release, I cannot afford to analyze the same problem twice ...

infradom, 2025-08-28:
Sorry, the more recent releases were only published 6 hours ago, my apologies for that last remark

infradom, 2025-08-28:
I believe this is a very good idea, but we would need to probe each individual  (enabled or required) register, probing in blocks is difficult since if a block fails, we need to split the block anyway. 
I am indeed tired of searching for a register that all of a sudden does not work anymore ... ``
The startup delay might be 3 to 10 seconds in my opinion, but it is worth it ...

infradom, 2025-08-28:
- Question 1: the changes are huge: disabled sensor registers are not read anymore and we had to implement a depencency chain between select, number and swith and their corresponding sensor. 
I absoletly do not understand why this one failing modbus read did work in previous versions. My guess is that it failed in the previous version also but that the ignore_readerror mechanism is broken during one of the changes since beginning of july

- Question 2: this is a standard HA question that is not related to this integration. The available history mechanisms apply to all integrations. I personally sent my history to influxdb with a grafana frontend. But even without influxdb, I can look back into the history for each entity.

edli83, 2025-08-28:
thank´s for clarifying

edli83, 2025-08-28:
Sounds good, thank you very much for your work.
I hope you will be able to implement soon. So long I can stay with 2025.07.2 - no problem.
Good luck and not so many issues  ... :-)

rosenrot00, 2025-08-28:
@infradom I opened one here https://github.com/wills106/homeassistant-solax-modbus/issues/1570. I think we could do all registers, if fails, half, the half the works, mark as "working", the other, split again,........

Let me think more than 5 minutes about it and move the discussion there - as suggested.