wills106/homeassistant-solax-modbus #1548: Solax - After update to 2025.8.1/2 values are updated very slow, almost every minute.

itrankolov, 2025-08-20:
I also have this in HA logs:

Logger: custom_components.solax_modbus
Source: custom_components/solax_modbus/__init__.py:829
integration: SolaX Inverter Modbus ([documentation](https://homeassistant-solax-modbus.readthedocs.io/), [issues](https://github.com/wills106/homsassistant-solax-modbus/issues))
First occurred: 10:40:43 AM (2 occurrences)
Last logged: 10:41:33 AM

Something went wrong reading from modbus: Attribute hass is None for <entity unknown.unknown=unknown>
Traceback (most recent call last):
  File "/config/custom_components/solax_modbus/__init__.py", line 829, in async_read_modbus_data
    res = await self.async_read_modbus_registers_all(group)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/solax_modbus/__init__.py", line 1056, in async_read_modbus_registers_all
    if sens and not descr.internal: sens.modbus_data_updated() # publish state to GUI and automations faster
                                    ~~~~~~~~~~~~~~~~~~~~~~~~^^
  File "/config/custom_components/solax_modbus/sensor.py", line 249, in modbus_data_updated
    self.async_write_ha_state()
    ~~~~~~~~~~~~~~~~~~~~~~~~~^^
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 1023, in async_write_ha_state
    self._async_verify_state_writable()
    ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^^
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 987, in _async_verify_state_writable
    raise RuntimeError(f"Attribute hass is None for {self}")
RuntimeError: Attribute hass is None for <entity unknown.unknown=unknown>

infradom, 2025-08-21:
Thanks for the logfile. It will certainly help - I need some time since there are a couple of other bugs requiring attention .. 
Then I will upgrade my HA to the same version you have and see if the error then also occurs on my system.

itrankolov, 2025-08-21:
If it will be in any help - I can report, that the issue is not present with Solax X3-Hybrid-15kW and X3-Ultra-20kW
 with the same HA and integration versions (2025.8.2), but it is another HA instance, so it may be related to my HA install or inverter firmware version, but I have no idea how to check that.

infradom, 2025-08-21:
I just upgraded my system to HA 2025.8.2 and cannot reproduce your problem on my system.
The error in the logs looks very strange, at the time of writing, I have no direct clue what is causing this ...

infradom, 2025-08-21:
> By "test with a soruce file" do you mean changing some file in here: /config/custom_components/solax_modbus ?

yes, this is what I meant, I would tell you which lines to change

gernotgernot, 2025-08-21:
> In my case I figured out there is an issue because HA updates values every 1m or so, not 5s. Can you check the update rate of your sensors ?

i have some data that shows 12 seconds interval (my normal interval), but also some times very i only see updates every 45 to 120 sconds..... might be the sleep mode that infradom mentioned.

gernotgernot, 2025-08-21:
ok, i have added the line and restartet homeassistant. error is back. can you please help me how to create the full debug of the HACS solax modbus integration?

infradom, 2025-08-21:
Finding a solution is not as easy as I thought, it is unclear why this computed entity performs differently than the other ones. Still searching ...
I also dont understand why this does not happen on all systems.

gernotgernot, 2025-08-21:
Thanks for your work, if you can do anything to help please ask.
additional information if needed:

Home Assistant

Installation method
Home Assistant OS
Core
2025.8.2
Supervisor
2025.08.1
Operating System
16.0
Frontend
20250811.0

Inverter: 

Device info
X3-Hybrid-15kW
by SolaX Power
Firmware: DSP v1.51 ARM v1.49
Hardware: Gen4

itrankolov, 2025-08-21:
I've tried it also, it still gives the error and is still slow to update.

itrankolov, 2025-08-21:
Here is the log after enabling the sensor (temp fix is still in place).

[home-assistant_solax_modbus_2025-08-21T14-33-21.998Z.log](https://github.com/user-attachments/files/21919886/home-assistant_solax_modbus_2025-08-21T14-33-21.998Z.log)

itrankolov, 2025-08-21:
Look in the hidden sensors, or look for it from entities.

infradom, 2025-08-21:
Thanks for testing, but have you tested the fix as published in the PR:
- one additional import line 59 
- one change in line 1055

Or have you tested the previous temporary fix (that did not work)

```
from .sensor import SolaXModbusSensor, is_entity_enabled
```
and 

```
            if sens and is_entity_enabled(self._hass, self, descr) and not descr.internal: sens.modbus_data_updated() # publish state to GUI and automations faster
```

infradom, 2025-08-21:
Sorry indeed there is an error in the solution ...

itrankolov, 2025-08-21:
I tested the one in the PR :)

itrankolov, 2025-08-21:
Applied the patch to sensor.py as well and it seems to be working as it should. Thanks!