wills106/homeassistant-solax-modbus #1549: [Bug]:  After updating from version 2025.6.2 to version 2025.8.1/2, the value in the number.solax_export_control_user_limit entity is interpreted incorrectly.

George-OPA, 2025-08-20:
### Describe the bug

After the update, the decimal point is incorrectly shifted for this entity. For example, setting HA to 300W causes the actual setting in the inverter (export limitation) to be 3000W !!!

### Integration Version

2025.08.2

### Homeassistant core version

2025.8.2

### Inverter brand

X3-Hybrid-15kW

### Plugin used

plugin_solax.py

### Serial prefix

H34A15

### Inverter firmware versions

DSP v1.55 ARM v1.51

### Connection Method

WiFi+LAN

### Dongle firmware

FW 1.006.01

George-OPA, 2025-08-21:
Yesterday, I quickly reverted to version 2025.07.7, which solved the problem. Today, after upgrading to 2025.08.2 and correcting number.config_export_control_limit_readscale to (10), everything seems to be OK. Thank you very much 👍

infradom, 2025-08-21:
Thanks for the feedback !