wills106/homeassistant-solax-modbus #1562: [Bug]: X3-HYB-G4 PRO - Wrong value for EPS Yield Total

InToSSH, 2025-08-26:
### Describe the bug

Hello,
just installed the X3-HYB-G4 PRO 15kW, connection went smoothly but the `EPS Yield Total` sensor seems to be way off.
It is showing that today it ran through almost 2MWh of energy.

I am  running the 2025.08.3 version, not the latest, but checking the changelog it doesn't seem to mention any of these issues.

[image]

As a comparison, the `EPS Yield Today` seems to show correct values:

[image]

---
There also might be some discrepancy between the `Today's Solar Energy` and `Battery Input Energy Today`.
It is showing that today the Solar generated 15kWh, but I pushed 23.7kWh into the battery. Keep in mind that there is currently no grid connected, so the battery couldn't have been charged from the grid, meaning the Solar generated should be higher that the energy put into the battery.

[image]

[image]

But I will monitor these tomorrow to see if it settles.

### Integration Version

2025.08.3

### Homeassistant core version

2025.8.3

### Inverter brand

SolaX Power

### Plugin used

plugin_solax.py

### Serial prefix

10K0F1

### Inverter firmware versions

ARM 2.03, DSP 2.07, Internal code: 009.01 009.01 06

### Connection Method

Solax Pocket Wifi+LAN V3.0

### Dongle firmware

1.005.04

rosenrot00, 2025-08-27:
Regarding the second point, this is my opinion: It’s not a coding bug but how the inverter registers are defined. Solar Today only counts the PV energy converted to AC, while Battery Input Today counts the DC energy charged into the battery. To get the actual PV harvest, you need to add both values together, plus a small margin for conversion losses.

infradom, 2025-08-27:
The pro is rather new, so we indeed gradually find the correct config of our integration.
If we are sure that all pro's have an incorrect scaling, we can adapt the scaling factor for that entity. Otherwise it gets ugly and apply scaling exceptions based on serial number prefix (based on software version is more difficult to implement)

Probably non related: Please note that there is a disabled entity that let you control the export_control_limit_readscale (set to either 0.1, 1.0 or 10.0) . But your issue is not (only) related to that parameter.

infradom, 2025-08-27:
I checked the manual, and EPS total yield is expressed in 0.1kWh units. So if the pro is classified as GEN6, the source code is correct and complies to the documentation.

wills106, 2025-08-27:
It could be a typo in the docs. They are the first version and look to be from 2024, although that doesn't make sense how new the Inverters are.

It could also be possible that X1 ie the Vast needs a scale of 0.1 and the X3 Pro G4 0.01 or times 10 etc to be correct?

Like I mentioned in #1380 I had to guess a lot of the registers and the docs are missing quite a bit of information for this Inverter.

InToSSH, 2025-08-27:
@infradom Tried setting the `export_control_limit_readscale` as the `export_control_user_limit` was off. I had to set it to 10 to make it read the same value as reported directly from the Solax inverter. The issue here is that the entity is capped to 0-2500W so now it reads correctly in W but I can't go higher than 2500W. I have to set the scale to 1 and set the value to 1000W to have 10kW export limit.

Regarding the Solar Total I feel like it should show the solar production alone, regardless if it is converted to AC or goes to battery. I will have to double check with my father-in-law and his G4. Either way the integration reads the same values as Solax Cloud directly, even though those values are quite meaningless and I would rather see the entire PV production. It feels like a FW bug.

[image]

@wills106 Thank you for looking into this, if you need any help debugging or testing the registers, I would be more than happy to help.

InToSSH, 2025-08-28:
@wills106 thank you and yes, the Battery 1 Temp is consistent with SolaxCloud.

wills106, 2025-08-28:
Added some more missing Registers [2025.08.6](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/2025.08.6)

wills106, 2025-08-28:
The scale was previously `scale=0.1,` now it's zero.
Does it need to go the other way then and become `scale=0.01,`?