wills106/homeassistant-solax-modbus #1563: [Bug]: 'Solis' - 'S5-EH1P3K-L'  Error doing job: Task exception was never retrieved (None)

chronvas, 2025-08-27:
### Describe the bug

In the error logs I see this issue:

This error originated from a custom integration.

Logger: homeassistant
Source: custom_components/solax_modbus/switch.py:101
integration: SolaX Inverter Modbus (documentation, issues)
First occurred: 5:58:01 PM (1 occurrence)
Last logged: 5:58:01 PM
```
Error doing job: Task exception was never retrieved (None)
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 1140, in _async_update_entity_states
    await asyncio.gather(*tasks)
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 981, in async_update_ha_state
    self._async_write_ha_state()
    ~~~~~~~~~~~~~~~~~~~~~~~~~~^^
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 1151, in _async_write_ha_state
    self.__async_calculate_state()
    ~~~~~~~~~~~~~~~~~~~~~~~~~~~~^^
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 1088, in __async_calculate_state
    state = self._stringify_state(available)
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 1032, in _stringify_state
    if (state := self.state) is None:
                 ^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 1659, in state
    if (is_on := self.is_on) is None:
                 ^^^^^^^^^^
  File "/config/custom_components/solax_modbus/switch.py", line 101, in is_on
    sensor_value = int(self._hub.data[self._sensor_key])
TypeError: int() argument must be a string, a bytes-like object or a real number, not 'NoneType'
```

### Integration Version

2025.08.5

### Homeassistant core version

2025.8.3

### Inverter brand

Solis

### Plugin used

plugin_solis.py

### Serial prefix

603101

### Connection Method

Elfin-EW1X + GC-1201S RS485 / Modbus Dual Host Adapter

### Additional context

Error appeared also in the previous version and it is persistent through system restarts. Unsure which version the error started from.
debug log file with startup info attached

[home-assistant_solax_modbus_2025-08-27T14-58-04.716Z.log](https://github.com/user-attachments/files/22009157/home-assistant_solax_modbus_2025-08-27T14-58-04.716Z.log)

infradom, 2025-08-27:
To understand better I will modify the code for switch.py's is_on property:

```

    @property
    def is_on(self):
        """Return the state of the switch."""
        # Prioritize user action within debounce time
        if self._last_command_time and datetime.now() - self._last_command_time < DEBOUNCE_TIME:
            return self._attr_is_on

        # Otherwise, return the sensor state
        if self._sensor_key and self._sensor_key in self._hub.data:
            sensvalue = self._hub.data.get(self._sensor_key, None)
            if sensvalue: sensor_value = int(sensvalue)
            else: 
                _LOGGER.error(f"{self._hub.name} Sensor {self._sensor_key} corresponding to switch {self._key} bit {self._bit} has no integer value {sensvalue}")
                sensor_value = 0 # probably completely wrong, but at least we can continue with other entities
            return bool(sensor_value & (1 << self._bit))

        return self._attr_is_on

```

This will not solve the problem, but the exception will not occure anymore and we will get an error message that displays the context so that we can find which sensor key has not value. If you are familiar with source code modification, you could test this, otherwise this enhanced code will be in a next release

chronvas, 2025-08-28:
@infradom Thank you so much for this!
I have just upgraded to `2025.08.7` and I get the following:
```
This error originated from a custom integration.

Logger: custom_components.solax_modbus.switch
Source: custom_components/solax_modbus/switch.py:104
integration: SolaX Inverter Modbus (documentation, issues)
First occurred: 6:03:46 PM (8 occurrences)
Last logged: 6:03:46 PM

solisinverter: Sensor special_settings corresponding to switch i_leak_protection bit 3 has no integer value None
solisinverter: Sensor special_settings corresponding to switch pv_iso_protection bit 4 has no integer value None
solisinverter: Sensor special_settings corresponding to switch grid_interference_protection bit 5 has no integer value None
solisinverter: Sensor special_settings corresponding to switch dc_component_of_grid_current_protection_switch bit 6 has no integer value None
solisinverter: Sensor special_settings corresponding to switch const_voltage_mode_enable bit 7 has no integer value None
```
Let me know if the full logs are needed and I will upload them as soon as I can. Thank you

infradom, 2025-08-28:
@chronvas : thanks a lot. From your older log, I could see that special_settings was read without modbus errors, but it is unclear why the value would be None.
A new log might certainly help (but I am not sure)

chronvas, 2025-08-28:
Here's a longer log, ty 

[home-assistant_solax_modbus_2025-08-28T16-13-24.730Z.log](https://github.com/user-attachments/files/22029222/home-assistant_solax_modbus_2025-08-28T16-13-24.730Z.log)

infradom, 2025-08-28:
I think I found the error:
line 100 in switch.py should become:
```
        if self._sensor_key and (self._sensor_key in self._hub.data):
```
So I believe the added brackets are needed.
I will issue a pull request, hopefully the new code can be released soon

Am I right when I say that most entities now have correct values, except the switches derived from the "special_settings" register ?

Your new log shows that the special_settings register could be read without errors, but when we need to extract the bits from this register, the register value is None ...

chronvas, 2025-08-28:
All values look correct to me (except the battery in/out from the other issue)
The spedial_settings looks like this 

[image]
The other "protection" values appear as switches so might be boolean?

infradom, 2025-08-28:
I added some more brackets in line 96 of switch.py. This time I believe they are really needed. There is hope 
```
        if self._last_command_time and ((datetime.now() - self._last_command_time) < DEBOUNCE_TIME):
```

Special_settings is an number that is just a collection of bits. So the bitflags are extracted from special_settings.
I am not the author of this piece of code in switch.py, I am trying to resolve the issue in my latert PR.
Hopefully the code can be released soon.

The battery input output should already be solved a  while ago. you have to enable the entities battery_power and battery_charge_direction as explained in the other solis bug thread.

New users do not suffer from this battery problem, only existing users need to enable these 2 entities. A final solution will come but I want to take the time to do it correctly.

infradom, 2025-08-30:
Battery input/output  problems should also be solved in the next release, the dependency mechanism is implemented.
Will soon be published. Closing