wills106/homeassistant-solax-modbus #1572: [Bug]: Strange behavior of Remote control - Enable Grid Control

George-OPA, 2025-08-28:
### Describe the bug

The error always appears after 10 to 20 minutes after turning on the controlled export from the grid view, "**Enable Grid Control**". The entities "sensor.solax_house_load" and "**sensor.solax_grid_export**" fluctuate, sometimes "**sensor.solax_house_load**" goes to negative values.
This error has been present for some time, and I am not sure if it is a firmware error. The error occurs exclusively in "Enable Grid Control" mode during controlled export. Please check if others are experiencing similar behavior. Thank you.

### Integration Version

2025.08.7

### Homeassistant core version

2025.8.3

### Inverter brand

X3-Hybrid-15kW

### Plugin used

plugin_solax.py

### Serial prefix

H34A15

### Inverter firmware versions

ARM 1.54-1.09 / DSP v1.56

### Connection Method

WiFi+LAN

### Dongle firmware

1.007.02

### Additional context

[image]

https://github.com/user-attachments/assets/6a2607f2-67b6-476e-82e5-d282c077a495

George-OPA, 2025-10-06:
Has anyone managed to replicate this state? Simulate it? For me, it's always replicable and it behaves more or less the same strangely.

infradom, 2025-10-06:
I tend to ignore bug reports without rull debug log (see general faq)

infradom, 2025-10-06:
Maybe you have heavily fluctuating loads. Some kitchen cooking gear switches continuously on/off every few seconds

closed 2025-11-22 by github-actions[bot]