wills106/homeassistant-solax-modbus #1574: [Bug]: SolxModbus not starting

WensleyStock, 2025-08-29:
### Describe the bug

Hi, Since 6th August 9I have been away) Solax modbus will not start.
Version 2024.12.2

### Integration Version

Version 2024.12.2

### Homeassistant core version

2025.8.3

### Inverter brand

Solax

### Plugin used

plugin_solax.py

### Serial prefix

H1E502

### Inverter firmware versions

ARM 3.36-1.05 DSP 3.43 Module Version 2.034.03

### Connection Method

No dongle, RJ45 cable connection to invertor

### Dongle firmware

No dongle

### Detailed Error Log

Logger: homeassistant.util.package
Source: util/package.py:159
First occurred: 1:18:36 PM (3 occurrences)
Last logged: 1:18:45 PM

Unable to install package pymodbus==3.6.9: × No solution found when resolving dependencies: ╰─▶ Because you require pymodbus==3.6.9 and pymodbus==3.9.2, we can conclude that your requirements are unsatisfiable.

Logger: homeassistant.setup
Source: setup.py:278
First occurred: 1:18:45 PM (1 occurrence)
Last logged: 1:18:45 PM

Setup failed for custom integration 'solax_modbus': Requirements for solax_modbus not found: ['pymodbus==3.6.9']

### Additional context

Has been working fine.  Failed whilst I was away.  Only HA core and OS updates applied whilst I was away as far as I can recall.

Any suggestions or help much appreciated.

Kars-de-Jong, 2025-08-29:
You're running quite an old version of this integration, have you tried upgrading to the latest release?

WensleyStock, 2025-08-29:
Thanks Kars. 

Update resolved.

Lesson learnt.

Thank you again.

wills106, 2025-08-30:
Closing as resolved.