wills106/homeassistant-solax-modbus #1578: Integration does not load any more after last update

RKeme, 2025-09-01:
### Describe the bug

Integration does not load any more after last update 

### Integration Version

2025.08.08

### Homeassistant core version

2025.8.3

### Inverter brand

Sofar

### Plugin used

plugin_sofar.py

### Serial prefix

062000

### Connection Method

LSE3

### Detailed Error Log

Logger: homeassistant.config_entries
Quelle: config_entries.py:749
Erstmals aufgetreten: 04:53:17 (2 Vorkommnisse)
Zuletzt protokolliert: 05:02:30

Error setting up entry Sofar for solax_modbus
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 749, in __async_setup_with_context
    result = await component.async_setup_entry(hass, self)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/solax_modbus/__init__.py", line 263, in async_setup_entry
    plugin = await hass.async_add_executor_job(_load_plugin, plugin_name)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.13/concurrent/futures/thread.py", line 59, in run
    result = self.fn(*self.args, **self.kwargs)
  File "/config/custom_components/solax_modbus/__init__.py", line 234, in _load_plugin
    plugin = importlib.import_module(f".plugin_{plugin_name}", "custom_components.solax_modbus")
  File "/usr/src/homeassistant/homeassistant/util/loop.py", line 201, in protected_loop_func
    return func(*args, **kwargs)
  File "/usr/local/lib/python3.13/importlib/__init__.py", line 88, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
           ~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "<frozen importlib._bootstrap>", line 1387, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1360, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1331, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 935, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 1026, in exec_module
  File "<frozen importlib._bootstrap>", line 488, in _call_with_frames_removed
  File "/config/custom_components/solax_modbus/plugin_sofar.py", line 228, in <module>
    SofarModbusButtonEntityDescription(
    ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^
        name="Passive: Update Battery Charge/Discharge",
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    ...<6 lines>...
        depends_on= ("passive_mode_grid_power", "passive_mode_battery_power_min", "passive_mode_battery_power_max",),
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    ),
    ^
TypeError: SofarModbusButtonEntityDescription.__init__() got an unexpected keyword argument 'depends_on'

infradom, 2025-09-01:
Oops, sorry  - correction will follow soon
Probably only growatt users will suffer from this

infradom, 2025-09-01:
Sorry, I was also working on a PR for this, we should also extend button.py in my opinion:

```
            if isinstance(deplist, str): deplist = (deplist, )
       
     if isinstance(deplist, (list, tuple,)):
                _LOGGER.debug(f"{hub.name}: {button_info.key} depends on entities {deplist}")
                for dep_on in deplist: # register inter-sensor dependencies (e.g. for value functions)
                    if dep_on != button_info.key: hub.entity_dependencies.setdefault(dep_on, []).append(button_info.key) # can be more than one

    async_add_entities(entities)
```

rosenrot00, 2025-09-01:
Please go ahead. I will remove the mentioned PR.

RKeme, 2025-09-01:
Works for me - thx

closed 2025-09-01 by RKeme