wills106/homeassistant-solax-modbus #1582: [Bug]: X3 G4 PRO - cannot edit battery current over 20

ripperbla, 2025-09-01:
### Describe the bug

hi,
after changing max battery charging current from 25A to 20A I am not able to change it over 20A.

### Integration Version

2025.08.8

### Homeassistant core version

2025.8.2

### Inverter brand

Solax

### Plugin used

plugin_solax.py

### Serial prefix

10K0A1

### Inverter firmware versions

ARM 015.03-000.06, DSP 014.01

### Connection Method

Wifi + LAN

### Dongle firmware

1.005.04

### Detailed Error Log

[image]

[image]

wills106, 2025-09-01:
It will need an entry here:
https://github.com/wills106/homeassistant-solax-modbus/blob/e6a7846bc1d665e2518890c0c2e4ae70b2910bce/custom_components/solax_modbus/plugin_solax.py#L486

Also here:
https://github.com/wills106/homeassistant-solax-modbus/blob/e6a7846bc1d665e2518890c0c2e4ae70b2910bce/custom_components/solax_modbus/plugin_solax.py#L526

What are the actual limits of this Inverter?

ripperbla, 2025-09-01:
Max charging/discharging limit is 30A.

MAX_CURRENTS = [
    ("10K", 30), #Gen6 X3 G4 PRO

MAX_EXPORT = [
    ("10K0A", 11000), #X3 G4 PRO 10kW

ripperbla, 2025-09-01:
When I have edited it locally, it works.

[image]

wills106, 2025-09-02:
Just released [2025.09.2](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/2025.09.2) with the MAX_CURRENTS set to 25A for now. Are you sure you can use 30A as I looked at the data sheet on SolaX website and it mentions 25A not 30A?

I haven't added in the export limit yet, as will probably need to do it dynamically after converting the Hex in to decimal like:
https://github.com/wills106/homeassistant-solax-modbus/blob/2dfd7d495d9261e7dd71111640b12efb9a61c69a/custom_components/solax_modbus/plugin_solax.py#L8564-L8565

closed 2025-09-04 by ripperbla