wills106/homeassistant-solax-modbus #1590: [FR] Support SolaX X3-Aelio

wills106, 2025-09-02:
Hi @wills106, I was able to successfully load the Aelio with your instructions, but from the sensors' perspective the Modbus table seems to be different. I see the main differences between MTTP 5-6 (Aelio 50kW has 5 MPPTs and 60kW has 6 MPPTs). What is not loading correctly is the total PV count, the current SOC shows 0%. I can give you access to HA, but it is waiting for a public IP address. Cloud is not enabled yet.
Another difference was that when I registered Aelio to the Solax cloud as resident, I could not change the export control (item is not under remote control settings), but when I moved it to C&I, I could change the export control to "0", so the configuration is different. See screenshot.
[image]

Still no response from Solax regarding the Aelio Modbus table. :-(

Aelio - Solax Modbus screens
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

_Originally posted by @butvoo in https://github.com/wills106/homeassistant-solax-modbus/discussions/1555#discussioncomment-14286798_

wills106, 2025-09-05:
You can send me a message on https://community.home-assistant.io/u/wills106 same username. Then you can email it to me if you wish.

butvoo, 2025-10-22:
> > Only open items is how to handle different scale...?
> > name="Battery 1 Power Charge"
> > scale=10,
> 
> Just thought we can use the scale altering mechanism based off the Serial:
> 
> [homeassistant-solax-modbus/custom_components/solax_modbus/sensor.py](https://github.com/wills106/homeassistant-solax-modbus/blob/48f5d71c549a01f5625403ffcb35cf0bf0ecf530/custom_components/solax_modbus/sensor.py#L177-L180)
> 
> Lines 177 to 180 in [48f5d71](/wills106/homeassistant-solax-modbus/commit/48f5d71c549a01f5625403ffcb35cf0bf0ecf530)
>  def entityToListSingle(hub, hub_name, entities, groups, computedRegs, device_info: DeviceInfo, newdescr, readPreparation, readFollowUp):  # noqa: D103 
>      if newdescr.read_scale_exceptions: 
>          for (prefix, value,) in newdescr.read_scale_exceptions: 
>              if hub.seriesnumber.startswith(prefix):  newdescr = replace(newdescr, read_scale = value) 
> 
> Like we do for export control.

So, can you release it and I can test or what is the next step?

peterbomers, 2026-02-05:
@butvoo can you share the exact configuration you did to connect the Aelio modbus?
Im trying to connect the 60kW version, but luck so far. 
Thank you in advance.

butvoo, 2026-03-01:
@wills106 Could you please help me with Solax Aelio again?
The Export Control User Limit still shows "0", but when I set the Export Control User Limit to "1400", it changes to "14" in Solax Cloud Export Control, which is in KWatt. What could be wrong that the integration still shows "0"? write works.

[image]