wills106/homeassistant-solax-modbus #1594: [Bug]: Integration not load

Saentist, 2025-09-03:
### Describe the bug

Integration not load

[image]

### Integration Version

2025.09.4

### Homeassistant core version

2025.8.3

### Inverter brand

Growatt

### Plugin used

plugin_growatt.py

### Serial prefix

AL1.0ZABA

### Connection Method

https://github.com/rosenrot00/esphome_modbus_bridge

### Detailed Error Log

```verilog
Logger: homeassistant.config_entries
Източник: config_entries.py:749
Възникнал за първи път: 12:27:12 (2 occurrences)
Последно регистриран: 12:27:12

Error setting up entry netgro for solax_modbus
Error setting up entry 4200esp for solax_modbus
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 749, in __async_setup_with_context
    result = await component.async_setup_entry(hass, self)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/solax_modbus/__init__.py", line 248, in async_setup_entry
    plugin = await hass.async_add_executor_job(_load_plugin, plugin_name)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.13/concurrent/futures/thread.py", line 59, in run
    result = self.fn(*self.args, **self.kwargs)
  File "/config/custom_components/solax_modbus/__init__.py", line 219, in _load_plugin
    plugin = importlib.import_module(f".plugin_{plugin_name}", "custom_components.solax_modbus")
  File "/usr/src/homeassistant/homeassistant/util/loop.py", line 201, in protected_loop_func
    return func(*args, **kwargs)
  File "/usr/local/lib/python3.13/importlib/__init__.py", line 88, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
           ~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "<frozen importlib._bootstrap>", line 1387, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1360, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1331, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 935, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 1026, in exec_module
  File "<frozen importlib._bootstrap>", line 488, in _call_with_frames_removed
  File "/config/custom_components/solax_modbus/plugin_growatt.py", line 89, in <module>
    class GrowattModbusSensorEntityDescription(BaseModbusSensorEntityDescription):
    ...<5 lines>...
        register_type: int = REG_HOLDING
  File "/config/custom_components/solax_modbus/plugin_growatt.py", line 92, in GrowattModbusSensorEntityDescription
    order16: int = Endian.BIG
                   ^^^^^^
NameError: name 'Endian' is not defined
```

rosenrot00, 2025-09-03:
@Saentist please let us know if the latest release fixes your issues. We mainly tested on Solax, so anything not working on Growatt is very welcome.

rosenrot00, 2025-09-03:
@Saentist could you share debug logs as well? They would help us understand what the issue is.

rosenrot00, 2025-09-03:
I see in the log there are a lot of messages regarding connection issues. Not sure all this is related to our changes.

Saentist, 2025-09-03:
@wills106 in your integration serial and firmwares are inverted 
[image]

user86000, 2025-09-03:
Solax 2025.9.4 and 2025.9.5 no data for inverter with solinteg.py. All entities unknown.
No error message.

Neighter with ha 2025.8.3 nor 2025.9.0
2025.9.3 works with both ha versions

[image]

rosenrot00, 2025-09-03:
> Solax 2025.9.4 and 2025.9.5 no data for inverter with solinteg.py. All entities unknown. No error message.
> 
> Neighter with ha 2025.8.3 nor 2025.9.0 2025.9.3 works with both ha versions
> 
> <img alt="Image" width="732" height="167" src="https://private-user-images.githubusercontent.com/122205057/485300395-901962e5-ce56-48cc-89ba-c7d2885351bc.png?jwt=eyJ0eXAiOiJKV1QiLCJhbGciOiJIUzI1NiJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NTY5MjgyOTIsIm5iZiI6MTc1NjkyNzk5MiwicGF0aCI6Ii8xMjIyMDUwNTcvNDg1MzAwMzk1LTkwMTk2MmU1LWNlNTYtNDhjYy04OWJhLWM3ZDI4ODUzNTFiYy5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjUwOTAzJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI1MDkwM1QxOTMzMTJaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT0yM2RiMDNkY2I3MDM1YWM5ZWRjMDhkMWYyNWI2ZWM0MDgwZDVkN2Q5OTdmZTk0ZWExOGQ3MWRkMzI1ODFiMjdjJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.gVA7UCd2P6V9Dd7XS6SukMQpQus7B4d8wOE_IMOad4E">

Could you provide a debug log by turning on debug output in the solax integration (3 dots) and then restart HA?

user86000, 2025-09-03:
> Could you provide a debug log by turning on debug output in the solax integration (3 dots) and then restart HA?

[home-assistant_solax_modbus_2025-09-03T20-19-29.586Z.log](https://github.com/user-attachments/files/22126630/home-assistant_solax_modbus_2025-09-03T20-19-29.586Z.log)

Saentist, 2025-09-04:
Core 2025.9.0 destroys lot of integration's with use **pymodbus** Endian

```py
ImportError: cannot import name 'Endian' from 'pymodbus.constants' (/usr/local/lib/python3.13/site-packages/pymodbus/constants.py)```

Saentist, 2025-09-04:
@rosenrot00 
After x.x.5 is ok
But after the HAOS update, pymodbus changes, broke other integrations with use it.
For example Growatt_modbus also crashed with endian error.