wills106/homeassistant-solax-modbus #1601: [Bug]: X3 G4 PRO - Reverting Battery Charge Upper SOC

ripperbla, 2025-09-04:
### Describe the bug

hi,
when I set value of battery SOC it reverts back in few seconds. I don't see any error

[image]

### Integration Version

2025.09.6

### Homeassistant core version

2025.9.0

### Inverter brand

Solax

### Plugin used

plugin_solax.py

### Serial prefix

10K0A1

### Inverter firmware versions

ARM 015.03-000.06, DSP 014.01

### Connection Method

Wifi + LAN

### Dongle firmware

1.005.04

### Detailed Error Log

2025-09-04 10:41:42.615 INFO (MainThread) [custom_components.solax_modbus.number] writing SolaX battery_charge_upper_soc number register 224 value 100 after div by readscale 1 scale 1 with mode 1

ripperbla, 2025-09-04:
if this is true, then yes

[image]

ripperbla, 2025-09-07:
Weird is that Selfuse min SOC works

ripperbla, 2025-09-07:
Can I help somehow? Test something?

infradom, 2025-09-24:
One additional note: 
In the protocol document, there is a note: "one way battery" in red for these 2 registers. Do you have multiple battery strings? That could explain things.

ripperbla, 2025-09-24:
hi,
yes, the value is changed in HA after change in Solax Cloud.
I have "one way battery". I can read the data from 0x010E, but can't write to 0x00E0.

ripperbla, 2025-09-25:
@rosenrot00 

C:\Users\lada\OneDrive\Python>python modbus_rw.py --host 10.30.1.111 --register 0x010e --count 2 --read
2025-09-25 15:39:57,239 DEBUG: Connection to Modbus server established. Socket ('10.20.1.202', 64940)
Reading Holding Registers (FC 0x03), Addr=270, Count=2
2025-09-25 15:39:57,349 DEBUG: Processing: 0x0 0x1 0x0 0x0 0x0 0x7 0x1 0x3 0x4 0x0 0x5a 0x0 0x1
2025-09-25 15:39:57,350 DEBUG: decoded PDU function_code(3 sub -1) -> ReadHoldingRegistersResponse(dev_id=0, transaction_id=0, address=0, count=0, bits=[], registers=[90, 1], status=1)
2025-09-25 15:39:57,350 DEBUG: Frame advanced, resetting header!!
2025-09-25 15:39:57,350 DEBUG: Raw register data: [90, 1]
Data (2 registers):
00 5A 00 01

C:\Users\lada\OneDrive\Python>python modbus_rw.py --host 10.30.1.111 --register 0x00E0 --value 80
2025-09-25 15:43:19,705 DEBUG: Connection to Modbus server established. Socket ('10.20.1.202', 65035)
Writing Register (FC 0x06), Addr=224, Value=80
2025-09-25 15:43:19,850 DEBUG: Processing: 0x0 0x1 0x0 0x0 0x0 0x3 0x1 0x86 0x4
2025-09-25 15:43:19,851 DEBUG: Frame advanced, resetting header!!
2025-09-25 15:43:19,851 DEBUG: Write response: ExceptionResponse(dev_id=1, function_code=134, exception_code=4)

writing does not work

rosenrot00, 2025-09-25:
Thanks for the specs. Could you try to write 0x005a or even 0x0049 to 0x00e0?

ripperbla, 2025-09-25:
@rosenrot00 

C:\Users\lada\OneDrive\Python>python modbus_rw.py --host 10.30.1.111 --register 0x00E0 --value 0x005a
2025-09-25 17:06:57,701 DEBUG: Connection to Modbus server established. Socket ('10.20.1.202', 51836)
Writing Register (FC 0x06), Addr=224, Value=90
2025-09-25 17:06:57,866 DEBUG: Processing: 0x0 0x1 0x0 0x0 0x0 0x3 0x1 0x86 0x4
2025-09-25 17:06:57,866 DEBUG: Frame advanced, resetting header!!
2025-09-25 17:06:57,866 DEBUG: Write response: ExceptionResponse(dev_id=1, function_code=134, exception_code=4)

C:\Users\lada\OneDrive\Python>python modbus_rw.py --host 10.30.1.111 --register 0x00E0 --value 0x0049
2025-09-25 17:07:33,306 DEBUG: Connection to Modbus server established. Socket ('10.20.1.202', 51848)
Writing Register (FC 0x06), Addr=224, Value=73
2025-09-25 17:07:33,466 DEBUG: Processing: 0x0 0x1 0x0 0x0 0x0 0x3 0x1 0x86 0x4
2025-09-25 17:07:33,467 DEBUG: Frame advanced, resetting header!!
2025-09-25 17:07:33,467 DEBUG: Write response: ExceptionResponse(dev_id=1, function_code=134, exception_code=4)

ripperbla, 2025-09-25:
OK, I wrote to them, let's see.

ripperbla, 2025-10-02:
[image]