wills106/homeassistant-solax-modbus #1602: [Bug]: TypeError: ModbusClientMixin.write_register() got an unexpected keyword argument 'slave'

hemetz, 2025-09-04:
### Describe the bug

with update to HASS Core 2025.9.0 and SolaX Inverter Modbus 2025.09.5 each command result in this error message:
[image]

### Integration Version

2025.09.5

### Homeassistant core version

2025.9.0

### Inverter brand

SolaX Power

### Plugin used

plugin_solax.py

### Serial prefix

H3BC30

### Inverter firmware versions

ARM 023.01 and DSP 021.03

### Connection Method

SolaX Pocket WiFi+LAN

### Dongle firmware

1.006.01

### Detailed Error Log

`2025-09-04 11:00:00.010 ERROR (MainThread) [homeassistant.components.automation.return2normal] 2_return to normal: Sequence action at step 1: Error executing script. Unexpected error for device at pos 1: ModbusClientMixin.write_register() got an unexpected keyword argument 'slave'
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/script.py", line 524, in _async_step
    await getattr(self, handler)()
  File "/usr/src/homeassistant/homeassistant/helpers/script.py", line 1029, in _async_step_device
    await device_action.async_call_action_from_config(
        self._hass, self._action, dict(self._variables), self._context
    )
  File "/usr/src/homeassistant/homeassistant/components/device_automation/action.py", line 72, in async_call_action_from_config
    await platform.async_call_action_from_config(hass, config, variables, context)
  File "/usr/src/homeassistant/homeassistant/components/select/device_action.py", line 121, in async_call_action_from_config
    await hass.services.async_call(
    ...<5 lines>...
    )
  File "/usr/src/homeassistant/homeassistant/core.py", line 2835, in async_call
    response_data = await coro
                    ^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/core.py", line 2878, in _execute_service
    return await target(service_call)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 850, in entity_service_call
    single_response = await _handle_entity_call(
                      ^^^^^^^^^^^^^^^^^^^^^^^^^^
        hass, entity, func, data, call.context
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    )
    ^
  File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 922, in _handle_entity_call
    result = await task
             ^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/select/__init__.py", line 188, in async_handle_select_option
[... 146 lines cut ...]
                    ^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/core.py", line 2878, in _execute_service
    return await target(service_call)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 850, in entity_service_call
    single_response = await _handle_entity_call(
                      ^^^^^^^^^^^^^^^^^^^^^^^^^^
        hass, entity, func, data, call.context
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    )
    ^
  File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 922, in _handle_entity_call
    result = await task
             ^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/select/__init__.py", line 188, in async_handle_select_option
    await self.async_select_option(option)
  File "/config/custom_components/solax_modbus/select.py", line 118, in async_select_option
    await self._hub.async_write_register(unit=self._modbus_addr, address=self._register, payload=payload)
  File "/config/custom_components/solax_modbus/__init__.py", line 729, in async_write_register
    return await self.async_lowlevel_write_register(unit, address, payload)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/solax_modbus/__init__.py", line 722, in async_lowlevel_write_register
    resp = await self._client.write_register(address, regs[0], **kwargs)
                 ~~~~~~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^
TypeError: ModbusClientMixin.write_register() got an unexpected keyword argument 'slave'`

rosenrot00, 2025-09-04:
@hemetz install the latest release and please report back if the issue persists. If not, please close this one.

hemetz, 2025-09-04:
@rosenrot00 issue resolved with release 2025.09.6

closed 2025-09-04 by hemetz