wills106/homeassistant-solax-modbus #1610: [Bug]: No infos after updating

josa42, 2025-09-05:
### Describe the bug

I have updated to Home Assistant 2024.09.1 and SolaX Inverter Modbus to 2025.09.8 now all sensors and controls are unavailable.

<br>

Before the update I had no issues. I tried downgrading SolaX Inverter Modbus, but that does not seem to have changed anything.

<br><br>

## Screenhots

[image]

[image]

[image]

### Integration Version

2025.09.8

### Homeassistant core version

2024.09.1

### Inverter brand

SolaX Power

### Plugin used

plugin_solax.py

### Serial prefix

H34A12

### Inverter firmware versions

DSP v1.43 ARM v1.42

### Connection Method

Solax Pocket Wifi 3.0

### Dongle firmware

1.002.09

### Detailed Error Log

[home-assistant_solax_modbus_2025-09-05T22-56-24.028Z.log](https://github.com/user-attachments/files/22182555/home-assistant_solax_modbus_2025-09-05T22-56-24.028Z.log)

pavel-B, 2025-09-06:
I concur. Same observation. Upon removal of device and readding it fails.

infradom, 2025-09-06:
Looking at the log, there is no communication at all, there must be something else wrong. Not a bug ...

wills106, 2025-09-06:
Have you tried [unplugging the PocketWiFi](https://homeassistant-solax-modbus.readthedocs.io/en/latest/solax-faq/#i-have-lost-my-entities-on-restart-update-using-pocketwifi-30)?

infradom, 2025-09-06:
Can you enable debug logging including HA restart as described in the readthedocs general FAQ ?
then we can see the debug lines that examine modbus version

wills106, 2025-09-06:
I have just updated to 2025.9.1
The sensors briefly went to unknown once HA was fully loaded and then carried on working as normal.

There was an update to the HA Modbus component in HA 2025.9.1, but it's still on pyModbus version 3.11.1 So nothing should have changed.

josa42, 2025-09-06:
Here is the debug log:

[home-assistant_solax_modbus_2025-09-06T06-39-39.097Z.log](https://github.com/user-attachments/files/22185012/home-assistant_solax_modbus_2025-09-06T06-39-39.097Z.log)

infradom, 2025-09-06:
@josa42 Thanks, looks like a perfect log, serial number ok, 
but I fear you have to wait longer after restart before generating the log, I am missing the second part with the polling
Can you redo this and wait at least 3 minutes after restart

josa42, 2025-09-06:
@infradom Oh wow, everything is back to normal and working now! No idea what caused this. I restarted mutliple times before. But I'll take it 🙂 

Thanks a lot @infradom and @wills106 for looking into this!

infradom, 2025-09-06:
The startup phase may take longer with this new version. We will look into this if an abnormal startup time could be the cause (maybe for both of you)

infradom, 2025-09-06:
Ok, I will close this, In case startup really takes too long, please send use a debug log including HA restart (see general FAQ)

closed 2025-09-06 by infradom