wills106/homeassistant-solax-modbus #1619: [Bug]: missing sensor.solax_remaining_battery_capacity

matomatusov, 2025-09-16:
### Describe the bug

Hello, 
do you know why the solax_remaining_battery_capacity sensor dropped out? I just noticed it now.

Thanks

[image]

### Integration Version

2023.09.9

### Homeassistant core version

2025.9.3

### Inverter brand

SolaX Power

### Plugin used

plugin_solax.py

### Serial prefix

H34A10

### Inverter firmware versions

 DSP v1.56 ARM v1.54

### Connection Method

Solax Packet Wifi 3+

### Dongle firmware

1007.02

matomatusov, 2025-09-16:
Yes, the sensor is still enabled. 
Where can I find debug logs? In configuration.yaml I have custom_components.solax_modbus: debug.

Thanks

rosenrot00, 2025-09-16:
Much simpler than that: https://github.com/wills106/homeassistant-solax-modbus/blob/0d59e63f4c4b8b70a041fe8f3a98c0e3fbbe0427/docs/faq.md

rosenrot00, 2025-09-17:
@infradom you are more familiar with the logs than I'm. I didn't see something catching my eye.

infradom, 2025-09-17:
According to the log file, the remaining_battery_capacity  0x118 was added to a  small block 0x116-118 and polled correctly (without visible errors).

I looked in the config, and it seems to be disabled by default: 

```    SolaXModbusSensorEntityDescription(
        name="Remaining Battery Capacity",
        key="remaining_battery_capacity",
        native_unit_of_measurement=UnitOfEnergy.WATT_HOUR,
        device_class=SensorDeviceClass.ENERGY_STORAGE,
        state_class=SensorStateClass.MEASUREMENT,
        entity_registry_enabled_default=False,
        register=0x118,
        register_type=REG_INPUT,
        unit=REGISTER_U32,
        allowedtypes=AC | HYBRID | GEN4 | GEN5,
```
I enabled it on my system through the web GUI, it appears, but the shown value is also 0
I never used this entity as battery_capacity is what I use (misleading name)

For our documentation:
In the logs, the modbus request is: (block 0x116-0x118):
``` 
send: 0x0 0x15 0x0 0x0 0x0 0x6 0x1 0x4 0x1 0x16 0x0 0x4
```

The modbus response seems to be:
```
recv: 0x0 0x15 0x0 0x0 0x0 0xb 0x1 0x4 0x8 0xd 0x80 0x0 0x0 0x0 0x0 0x0 0x0 extra data: 
2025-09-16 19:05:38.640 DEBUG (MainThread) [pymodbus.logging] Processing: 0x0 0x15 0x0 0x0 0x0 0xb 0x1 0x4 0x8 0xd 0x80 0x0 0x0 0x0 0x0 0x0 0x0
```

strange, either a firmware update changed the behavior, or we changed that entity (I cannot remember)
To be examined further ...

HA-Supi, 2025-10-02:
I guess that this bug is related to the firmware 1.54/1.56.
Can you try a downgrade? I also noticed with this firmware that battery balancing is not visible anymore. With firmware versions before [<1.50] you could see small amounts of watts with entity battery power charge after reaching 100% capacity.

closed 2025-11-02 by matomatusov